YosysHQ/yosys #3385: Undefiend Behaviour and Bad Allocation when parsing BLIF .names

ALGCDG, 2022-06-18:
## Steps to reproduce the issue

Consider the following BLIF file, with 31 inputs signals to a .names command.
```
.model test
.inputs w0 w1 w2 w3 w4 w5 w6 w7 w8 w9 w10 w11 w12 w13 w14 w15 w16 w17 w18 w19 w20 w21 w22 w23 w24 w25 w26 w27 w28 w29 w30
.outputs out
.names w0 w1 w2 w3 w4 w5 w6 w7 w8 w9 w10 w11 w12 w13 w14 w15 w16 w17 w18 w19 w20 w21 w22 w23 w24 w25 w26 w27 w28 w29 w30 out
1101010001100110010001100111001 0
.end
```
I have also attached it as a file:
[bug.blif.zip](https://github.com/YosysHQ/yosys/files/8933492/bug.blif.zip)

To recreate, please run: 
```
yosys -p "read_blif bug.blif" 
```

Interestingly, the issue also happens when the number of input signals is 31, 63, 95, 127, 159, 191... and so on (repeats every 32).
Possibly this is a result of how the compiler handles undefined behaviour when shifting an integer literal by more than 31.
Some compiler implementations sometimes handle this undefined behaviour by applying % 32 to shift argument, which might explain the pattern.

## Expected behavior

The BLIF parser already checks that the input plane is not over 12 bits when parsing the LUT.
Input plane size over 12 should be detected earlier while parsing the input signals of the .names command, and syntax error should be raised.
Perhaps it would be useful to have an error message which explains the LUT size limit of 12 bits.
Undefined behaviour of shifting more than 31 should be avoided.

## Actual behavior

When run with the current master version of Yosys, this produces the following failure:
```
yosys(51426,0x11706b5c0) malloc: can't allocate region
*** mach_vm_map(size=18446744071562067968) failed (error code=3)
yosys(51426,0x11706b5c0) malloc: *** set a breakpoint in malloc_error_break to debug
libc++abi.dylib: terminating with uncaught exception of type std::bad_alloc: std::bad_alloc
Abort trap: 6
```

Instrumenting Yosys with the Address Sanitizer confirms that this is a bad allocation during the RTLIL:Const constructor, which reserves a vector of the size of the LUT.

Address Sanitizer output:
```
==23177==WARNING: AddressSanitizer failed to allocate 0xffffffff80000000 bytes
==23177==AddressSanitizer's allocator is terminating the process instead of returning 0
==23177==If you don't like this behavior set allocator_may_return_null=1
==23177==AddressSanitizer CHECK failed: ../../../../src/libsanitizer/sanitizer_common/sanitizer_allocator.cc:218 "((0)) != (0)" (0x0, 0x0)
    #0 0x7fb7f4685bf2  (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xe9bf2)
    #1 0x7fb7f46a4575 in __sanitizer::CheckFailed(char const*, int, char const*, unsigned long long, unsigned long long) (/usr/lib/x86_64-linux-gnu/libasan.so.4+0x108575)
    #2 0x7fb7f468b332  (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xef332)
    #3 0x7fb7f45c4ff8  (/usr/lib/x86_64-linux-gnu/libasan.so.4+0x28ff8)
    #4 0x7fb7f467c40e in operator new(unsigned long) (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xe040e)
    #5 0x55e78fd19abf in std::vector<Yosys::RTLIL::State, std::allocator<Yosys::RTLIL::State> >::reserve(unsigned long) (/home/vagrant/yosys/yosys+0x12445abf)
    #6 0x55e78f89b006 in Yosys::RTLIL::Const::Const(Yosys::RTLIL::State, int) kernel/rtlil.cc:233
    #7 0x55e7918d1e93 in Yosys::parse_blif(Yosys::RTLIL::Design*, std::istream&, Yosys::RTLIL::IdString, bool, bool, bool) frontends/blif/blifparse.cc:517
    #8 0x55e79190f5ac in Yosys::BlifFrontend::execute(std::istream*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, Yosys::RTLIL::Design*) frontends/blif/blifparse.cc:644
    #9 0x55e78f6683e0 in Yosys::Frontend::execute(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, Yosys::RTLIL::Design*) kernel/register.cc:466
    #10 0x55e78f682566 in Yosys::Pass::call(Yosys::RTLIL::Design*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >) kernel/register.cc:316
    #11 0x55e78f6897fe in Yosys::Pass::call(Yosys::RTLIL::Design*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) kernel/register.cc:293
    #12 0x55e78ffb372c in Yosys::run_pass(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, Yosys::RTLIL::Design*) kernel/yosys.cc:1126
    #13 0x55e78f4d1d39 in main kernel/driver.cc:550
    #14 0x7fb7f1f49c86 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21c86)
    #15 0x55e78f4c6a69 in _start (/home/vagrant/yosys/yosys+0x11bf2a69)
```

Ravenslofty, 2022-06-21:
Fixed by #3375.

closed 2022-06-21 by Ravenslofty

ALGCDG, 2022-08-22:
I believe this bug is still an issue, will submit a test script and potential fix.

ALGCDG, 2022-09-17:
Hi @Ravenslofty, can we merge the fix? Need someone with write permission.

PR #3452 (2022-10-10): Add BLIF names command input plane size check
Adds a check while parsing BLIF names command that input plane has 12 or fewer input signals.
Makes consistent with later LUT size check and avoid undefined behaviour when calculating LUT size for names command.
Fixes #3385