YosysHQ/yosys #3400: Dual port RAM doesn't infer a DP16KD on ECP5 if both ports are on the same clock

robinsonb5, 2022-07-04:
## Steps to reproduce the issue

Here are two dual port RAM implementations, one using the same clock for both ports, one using separate clocks.

```
module dpram_singleclock
#(parameter DATA_WIDTH=8, parameter ADDR_WIDTH=6)
(
	input [(DATA_WIDTH-1):0] data_a, data_b,
	input [(ADDR_WIDTH-1):0] addr_a, addr_b,
	input we_a, we_b, clk,
	output reg [(DATA_WIDTH-1):0] q_a, q_b
);

	reg [DATA_WIDTH-1:0] ram[2**ADDR_WIDTH-1:0];

	// Port A 
	always @ (posedge clk)
	begin
		if (we_a) 
			ram[addr_a] <= data_a;
		else 
			q_a <= ram[addr_a];
	end 

	// Port B 
	always @ (posedge clk)
	begin
		if (we_b) 
			ram[addr_b] <= data_b;
		else 
			q_b <= ram[addr_b];
	end

endmodule

module dpram_dualclock
#(parameter DATA_WIDTH=8, parameter ADDR_WIDTH=6)
(
	input [(DATA_WIDTH-1):0] data_a, data_b,
	input [(ADDR_WIDTH-1):0] addr_a, addr_b,
	input we_a, we_b, clk, clk_b,
	output reg [(DATA_WIDTH-1):0] q_a, q_b
);

	reg [DATA_WIDTH-1:0] ram[2**ADDR_WIDTH-1:0];

	// Port A 
	always @ (posedge clk)
	begin
		if (we_a) 
			ram[addr_a] <= data_a;
		else 
			q_a <= ram[addr_a];
	end 

	// Port B 
	always @ (posedge clk_b)
	begin
		if (we_b) 
			ram[addr_b] <= data_b;
		else 
			q_b <= ram[addr_b];
	end

endmodule
```

```synth_ecp5 -top dpram_dualclock``` results in a DP16KD RAM block being inferred, whereas ```synth_ecp5 -top dpram_singleclock``` builds the RAM from logic cells instead.

Further, using the dualclock variant in a larger design which feeds both clk and clk_b from the same clock also causes the RAM to be built from logic rather than a RAM block.

## Expected behavior

I would expect a RAM block to be inferred whether or not the clock pins are fed from the same clock.

## Actual behavior

RAM block is inferred only if the two ports are fed from different clocks.

Ravenslofty, 2022-07-14:
What Yosys version are you using?

robinsonb5, 2022-07-14:
Current oss-cad-suite at the time of filing, so Yosys 0.18+48 (git sha1 c39bade1a, clang 10.0.0-4ubuntu1 -fPIC -Os)

e2kgh, 2022-08-27:
Any chance, somebody could look into that? Or any idea, of a work around?
THANKS!

robinsonb5, 2022-08-27:
Just to confirm this is still present as of today's oss-cad-suite:
Yosys 0.20+54 (git sha1 060cbd3e9, clang 10.0.0-4ubuntu1 -fPIC -Os)

mmicko, 2022-08-30:
Hi @robinsonb5 . Our latest memory inference implementation is trying to be as accurate as possible in implementation and simulations, and in this case it would require read-before-write semantic and for that reason it gets implemented as number of flip-flops when both clocks are same

There are two ways of making this work as you expect :
1. Add attribute ```(* no_rw_check *)``` before memory reg in dpram_singleclock module
2. Add additional parameter to synth_ecp5 command so something like ```synth_ecp5 -top dpram_singleclock -no-rw-check```

Hope this solves you issues.

robinsonb5, 2022-08-30:
Many thanks - that does indeed solve my issue.  So user-error, not in fact a bug - apologies for the noise.

closed 2022-08-30 by robinsonb5