YosysHQ/yosys #3405: Compiling Yosys from source tarball

titan73, 2022-07-08:
Hi,

I can't clone a github repo due to IT policy. I downloaded Yosys 0.19 and build it successfully but when it comes to building abc, Yosis makefile tries to download abc github repo when abc directory is not present which does not works.
Is it possible to add releases with tarballs also for Yosys's abc ?
Note that abc does not have releases either and I see Yosys's abc is 38 commit ahead from berkely's one so better using Yosy's one.

Thanks
Chris

mmicko, 2022-07-27:
Work now done as on master at https://github.com/YosysHQ/yosys/commit/bc012995b44ca9a68524ae3a1ecb4b871d1436c9 will be supported from next release

mmicko, 2022-08-03:
Now abc.tar.gz it is now part or release https://github.com/YosysHQ/yosys/releases/tag/yosys-0.20 
First unpack yosys and then unpack abc in underneath directory and build as usual.

closed 2022-08-03 by mmicko

titan73, 2022-08-03:
@mmicko Gunaxf! Jul qb lbh rapelcg gur zrfffntrf va gur znxrsvyr?

mmicko, 2022-08-03:
@titan73  :) messages are stored that way so it prevents grepping for messages to match when there is no actual error, but just executing that block of commands

titan73, 2022-08-07:
@mmicko Ah ok. Makes sense. It makes the makefile less readable though. Souldn't something like below more practical?
echo "ERR: ABC directory is ..." | sed 's/RR/RROR/'