YosysHQ/yosys #3411: There are something wrong with the yosys SCC algorithm?

1396510211, 2022-07-14:
## Steps to reproduce the issue

I encountered some problems when using the yosys to search the logic loop. When the logic loop consists of standard cells and user-defined cells, the "-all_cell_types" parameter seems to be normal. However, when the user-defined cells are not connected internally, The Yosys still incorrectly identifies the logic loop.

## Expected behavior
The logic loop can be correctly identified.
or How do I tell the difference between the two?

## Actual behavior
1、 a logic loop consists of standard cells and user-defined cells
```verilog
module TOP(input a,output b);

    wire d,c;
    assign d = a & c;

    TOP1 INST_TOP1(.x(d),.y(c));

endmodule

module TOP1(input x,output y);

    assign y = x;
    
endmodule

```
[image]

[image]
2、 a logic loop consists of standard cells and user-defined cells，but the user-defined cells are not connected internally
```verilog
module TOP(input a,output b);

    wire d,c;
    assign d = a & c;

    TOP1 INST_TOP1(.x(d),.y(c));

endmodule

module TOP1(input x,output y);

    
endmodule

```
[image]

[image]

Ravenslofty, 2022-07-14:
The `scc` pass does not look inside cells as it assumes a flat hierarchy (e.g. that `flatten` has been run before it). Running `flatten` replaces `INST_TOP1` with its contents, which results in `Z` and `in2` being disconnected, and there being no SCCs reported.

1396510211, 2022-07-15:
I see !

thanks !

closed 2022-07-15 by Ravenslofty