YosysHQ/yosys #3412: systh error

RanChenRivai, 2022-07-14:
When I try to synthesize my design, the run log shows the following :

##########################################################################################
OpenLane Container (b718fd0):/openlane$ ./flow.tcl -design alu -to yosys
OpenLane b718fd064063333703734ce3ae8bb56aa6810b0b
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using design configuration at /openlane/designs/alu/config.tcl
[INFO]: Sourcing Configurations from /openlane/designs/alu/config.tcl
[INFO]: PDKs root directory: /home/chenran/OpenLane-2/OpenLane/pdks
[INFO]: PDK: sky130A
[INFO]: Setting PDKPATH to /home/chenran/OpenLane-2/OpenLane/pdks/sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library is set to: sky130_fd_sc_hd
[INFO]: Sourcing Configurations from /openlane/designs/alu/config.tcl
[INFO]: Run Directory: /openlane/designs/alu/runs/RUN_2022.07.14_11.39.15
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Preparing LEF files for the min corner...
[INFO]: Preparing LEF files for the max corner...
[STEP 1]
[INFO]: Running Synthesis...
[ERROR]: during executing: "yosys -c /openlane/scripts/yosys/synth.tcl -l /openlane/designs/alu/runs/RUN_2022.07.14_11.39.15/logs/synthesis/1-synthesis.log |& tee /dev/null"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
[TCL: yosys -import] Command name collision: found pre-existing command `trace' -> skip.
Defining SYNTHESIS
Reading /openlane/designs/alu/runs/RUN_2022.07.14_11.39.15/tmp/synthesis/1-sky130_fd_sc_hd__tt_025C_1v80.no_pg.lib as a blackbox

1. Executing Liberty frontend.
Imported 428 cell types from liberty file.

2. Executing Verilog-2005 frontend: /openlane/designs/alu/src/rrv64_core_be_alu.sv
/openlane/designs/alu/src/rrv64_core_be_alu.sv:11: ERROR: syntax error, unexpected TOK_ID, expecting ',' or '=' or ')'
child process exited abnormally

[INFO]: Saving current set of views in 'designs/alu/runs/RUN_2022.07.14_11.39.15/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'designs/alu/runs/RUN_2022.07.14_11.39.15/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'designs/alu/runs/RUN_2022.07.14_11.39.15/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
###############################################################################################

So I go to  /openlane/designs/alu/src/rrv64_core_be_alu.sv:11  :
######################################
  1
  2 `include  "rrv64_core_param_pkg.svh"
  3 `include  "rrv64_core_typedef_pkg.svh"
  4
  5 module rrv64_core_be_alu
  6 (
  7     input                                   clk,
  8     input                                   rstn,
  9     input                                   input_valid,
 10     output                                  input_ready,
 11     input   rsvst_exe_comm_t     input_pld,
 12     output                                  int_reg_wr_valid,
 13     input                                   int_reg_wr_ready,
 14     output  [RRV64_INT_PHY_REG_ADDR_W-1:0]  int_reg_wr_addr,
 15     output  [RRV64_INT_REG_DATA_W-1:0]      int_reg_wr_data,
 16     output                                  rob_wr_vld,
 17     output  [RRV64_ROB_ADDR_W-1:0]          rob_wr_idx,
 18     output                                  idle
 19 );
##############################################

I confirm that this structure rsvst_exe_comm_t  has been defined in the header file, what is the reason for the synthesis failure?

Ravenslofty, 2022-07-14:
Did you call `read_verilog` with `-sv` to enable SystemVerilog extensions?

RanChenRivai, 2022-07-15:
Yes , I  callded 'read_verilog' with '-sv' in synth.tcl. Should I called it anywhere else？

RanChenRivai, 2022-07-15:
> Did you call `read_verilog` with `-sv` to enable SystemVerilog extensions?

Yes , I callded 'read_verilog' with '-sv' in synth.tcl. Should I called it anywhere else？

daglem, 2023-02-03:
> 
> I confirm that this structure rsvst_exe_comm_t has been defined in the header file, what is the reason for the synthesis failure?

rsvst_exe_comm_t would have to be a typedef for this to work. If it's defined as a variable, that would explain the problem.