YosysHQ/yosys #3416: nexus: memory_libmap pass produces unusable bitstream

danc86, 2022-07-19:
For the design I am working on, targetting Crosslink-NX17 (nexus family), the latest master branch of yosys (commit 08c319fc352fb2670b7416b5fb16ddcb9a400049) produces a bitstream that seems to be rejected by the device. I bisected the problem to commit 0a8eaca322a622610efe5a0d33d1cda83e4afa8c, which introduced the new memory_libmap pass for nexus.

The design uses explicit DPSC512K instances (LRAMs) and inferred PDPSC16K instances (EBRAMs). It seems like the memory_libmap pass is doing something invalid with those.

## Steps to reproduce the issue

1. Unpack [yosys-memory_libmap-bug-repro.zip](https://github.com/YosysHQ/yosys/files/9137249/yosys-memory_libmap-bug-repro.zip) into a new directory
2. `yosys hps_platform.ys`

Sorry I haven't minimized this reproducer yet, but just comparing the output between the two versions of yosys, bad (master branch) and good (master branch + revert of commit 0a8eaca322a622610efe5a0d33d1cda83e4afa8c), might be enough to find the problem.

## Expected behavior

Resulting bitstream should work :-)

## Actual behavior

nextpnr-nexus accepts the design and produces a bitstream successfully but it is rejected by the device.

danc86, 2022-07-19:
@tmichalak would Antmicro be able to help solve this issue?

Comparing the Verilog output from yosys in the good and bad cases, it seems the DPSC512K instances are the same in both cases, so it seems like the problem is more likely with the inferred PDPSC16K instances. I see that they are arranged and wired up quite differently between the good and bad cases. I wasn't able to spot anything immediately wrong but it might be worth a closer look.

gatecat, 2022-07-19:
This could well be an oxide or nextpnr issue too, I will try and have a look as well.

tmichalak, 2022-07-19:
@danc86 sure, we can take a look.

gatecat, 2022-07-19:
See my WIP in #3417

danc86, 2022-08-15:
I wanted to try the fixes from #3420 but it seems to cause an assertion failure in nextpnr. I'm testing with latest master branch: yosys c26b2bf543a226e65a3fb07040bb278d668accf2, nextpnr b653e39991429dc3350447f8b69030c454dd2ba7, prjoxide b6bd4bd33cacef4b7b0ff8a4137ee7e441b6bc15.

```
libc++abi: terminating with uncaught exception of type nextpnr_nexus::assertion_failure: Assertion failure: cell->tmg_index != -1 (/var/tmp/portage/sci-electronics/nextpnr-0.3_p20220811/work/nextpnr-b653e39991429dc3350447f8b69030c454dd2ba7/nexus/pack.cc:2629)
```

Should I file a bug against nextpnr, or did I miss a piece somewhere?

gatecat, 2022-08-15:
nextpnr, this looks like an unrelated issue from updating nextpnr 

just to check, did you rebuild prjoxide before rebuilding nextpnr?

danc86, 2022-08-19:
Sorry you're right, the above assertion failure was because I built nextpnr against old prjoxide. I rebuilt them in the correct order and that assertion failure is gone.

I think the fixes in #3420 might not be sufficient to make everything work here though. It still produces a non-working bitstream. I'll try to compare the latest yosys output vs. before memory_libmap to see if I can narrow down where the remaining problems are.

danc86, 2022-08-19:
Looking at the differences in the yosys-generated verilog before and after, the two things that stand out to me are:

* Many `DPR16X4` instances (distributed RAMs) have been replaced by a few `PDPSC16K` instances (EBRAMs). Not sure if this is expected/desirable or not. I guess it can make a difference to routability of the design depending on where the distributed RAMs are vs. the EBRAMs.
* The `PDPSC16K` instances now all have `.RESETMODE("ASYNC")` whereas before they didn't. `SYNC` is the default. I'm not clear on the consequences of this but it seems unexpected -- I can't ever figure out where in yosys this is coming from.

gatecat, 2022-09-16:
Big apologies for the delay (in part due to hardware in the wrong country...). I did some more investigation today, I think the following patch might be (part of?) a fix at least for a litex-boards SoC with the default configuration. Clocking off fairly soon but I'll be back on this next week.

```patch
diff --git a/techlibs/nexus/brams_map.v b/techlibs/nexus/brams_map.v
index 17306cded..cec56cec4 100644
--- a/techlibs/nexus/brams_map.v
+++ b/techlibs/nexus/brams_map.v
@@ -310,7 +310,7 @@ PDPSC16K #(
 ) _TECHMAP_REPLACE_ (
        .CLK(CLK_C),
 
-       .CEW(PORT_W_CLK_EN),
+       .CEW(PORT_W_CLK_EN & (|PORT_W_WR_EN)),
        .CSW(3'b111),
        .ADW(ADW),
        .DI(DI),
@@ -400,7 +400,7 @@ PDP16K #(
        .GSR("DISABLED"),
 ) _TECHMAP_REPLACE_ (
        .CLKW(PORT_W_CLK),
-       .CEW(PORT_W_CLK_EN),
+       .CEW(PORT_W_CLK_EN & (|PORT_W_WR_EN)),
        .CSW(3'b111),
        .ADW(ADW),
        .DI(DI),
```

SylvainLesne, 2022-10-28:
Hi, my apologies if this is off-topic, but I also stumbled upon this issue while using a LiteX SoC on a crosslink-NX device.

The above patch fixes the yosys+nextpnr flow for me, but I believe that there is another problem in the BRAM handling: it does not work with the yosys+radiant P&R flow.

As far as I understand, this seems related to the `CSDECODE_W/R` parameters. In bram_map.v there is `CSDECODE_?` set to `"111"`, and `CS?` set to `3'b111`. However, the NX primitive library (only available in the offline help of Radiant?) states: "Write chip select active setting / 0: active high / 1: active low", which is the opposite of what previous Lattice families did...
Maybe this could explain a difference of behavior between nextpnr and radiant P&R (and thus, an issue with the yosys+radiant flow)?