YosysHQ/yosys #3438: Synthesis error in Windows 11 and synthesis command error in WSL

tsp6, 2022-08-08:
1. First case in windows 11:

 I am using a windows 11 PC, I installed yosys toolchain to test in windows11.  Here below you can see that It can read a Verilog filer but I am facing an error in synthesizing step. the error says "Found control character or space (0x20) in string '$proc$D:\FPGA QNN\cc-toolchain-win\bin\yosys\share\techmap.v:106$25' which is not allowed in RTLIL identifiers". I opened techmap.v file to check the unwanted space, but it is very difficult to understand. Can you help me with this?

Terminal Output:

yosys> read -sv adder.v

1. Executing Verilog-2005 frontend: adder.v
Parsing SystemVerilog input from `adder.v' to AST representation.
Storing AST representation for module `$abstract\adder'.
Successfully finished Verilog frontend.

yosys> synth_gatemate -top adder

2. Executing SYNTH_GATEMATE pass.

2.1. Executing Verilog-2005 frontend: D:\FPGA QNN\cc-toolchain-win\bin\yosys\share\gatemate/cells_sim.v
Parsing Verilog input from `D:\FPGA QNN\cc-toolchain-win\bin\yosys\share\gatemate/cells_sim.v' to AST representation.

Synthesis error;

2.32.5. Executing OPT_MERGE pass (detect identical cells).
Finding identical cells in module `\adder'.
Removed a total of 0 cells.

2.32.6. Executing OPT_DFF pass (perform DFF optimizations).

2.32.7. Executing OPT_CLEAN pass (remove unused cells and wires).
Finding unused cells or wires in module \adder..

2.32.8. Executing OPT_EXPR pass (perform const folding).
Optimizing module adder.

2.32.9. Finished OPT passes. (There is nothing left to do.)

2.33. Executing TECHMAP pass (map to technology primitives).

2.33.1. Executing Verilog-2005 frontend: D:\FPGA QNN\cc-toolchain-win\bin\yosys\share\techmap.v
Parsing Verilog input from `D:\FPGA QNN\cc-toolchain-win\bin\yosys\share\techmap.v' to AST representation.
Generating RTLIL representation for module `\_90_simplemap_bool_ops'.
Generating RTLIL representation for module `\_90_simplemap_reduce_ops'.
Generating RTLIL representation for module `\_90_simplemap_logic_ops'.
Generating RTLIL representation for module `\_90_simplemap_compare_ops'.
Generating RTLIL representation for module `\_90_simplemap_various'.
Generating RTLIL representation for module `\_90_simplemap_registers'.
Generating RTLIL representation for module `\_90_shift_ops_shr_shl_sshl_sshr'.
ERROR: Found control character or space (0x20) in string '$proc$D:\FPGA QNN\cc-toolchain-win\bin\yosys\share\techmap.v:106$25' which is not allowed in RTLIL identifiers

[process exited with code 1 (0x00000001)]

2. Second case in wsl:
Now I tried running yosys toolchain in windows supported Linux system. Here I can read the Verilog file. But after that while running the synthesis using the command synth_gatemate -top adder I am facing the error " No such command: synth_gatemate ". if this is the case, can you help me with any kind of support to solve these issues?

Terminal Output:

yosys> read -sv adder.v
1. Executing Verilog-2005 frontend: adder.v
Parsing SystemVerilog input from `adder.v' to AST representation.
Generating RTLIL representation for module `\adder'.
Successfully finished Verilog frontend.

yosys> synth_gatemate -top adder
ERROR: No such command: synth_gatemate (type 'help' for a command overview)

yosys> help synth_gatemate
No such command or cell type: synth_gatemate

yosys> help synth_gatemate
No such command or cell type: synth_gatemate
[techmap.zip](https://github.com/YosysHQ/yosys/files/9280076/techmap.zip)
[adder.zip](https://github.com/YosysHQ/yosys/files/9280079/adder.zip)

nakengelhardt, 2022-08-08:
It looks like the problem is a space in your path. Can you rename the directory `D:\FPGA QNN` to `D:\FPGA_QNN` and see if that fixes the problem?

(This is still a techmap bug in escaping that we will investigate, but this workaround should let you work for now.)

tsp6, 2022-08-08:
Thank You @nakengelhardt , it worked.

closed 2022-08-08 by mmicko