YosysHQ/yosys #3440: wrong message about "use a ILP64 abi for larger tables"

maliberty, 2022-08-08:
## Steps to reproduce the issue

In https://github.com/YosysHQ/yosys/blob/60a787fa50062169c1aae4ecb199d13c0b6f25b6/kernel/hashlib.h#L191
```
	if (sizeof(int) == 4)
		throw std::length_error("hash table exceeded maximum size. use a ILP64 abi for larger tables.");
```
is not helpful.  ILP64 is quite dead and LP64 uses 4 byte ints.  If you want to allow larger values then you should use 'long long' not int as the type [long is 32bit on windows].    

A failure of this check is seen in https://github.com/The-OpenROAD-Project/OpenROAD/issues/2106

## Expected behavior

64 bit linux compiles (LP64) should allow for the larger table size.

## Actual behavior

64 bit linux compiles (LP64) don't allow for the larger table size and gives an unhelpful message.

closed 2022-08-25 by nakengelhardt

maliberty, 2022-10-10:
Why not use a longer type rather than just giving up?

nakengelhardt, 2022-10-10:
Using a longer type will involve an in-depth review of every single use of `int` in the entire codebase. Are you volunteering?

maliberty, 2022-10-10:
No but you closed this issue which suggests that yosys will not support larger designs.  It seems an important limitation of the tool to address.

nakengelhardt, 2022-10-10:
The issue as filed was that the error message was "wrong" (not true) and the suggestion it gave "not helpful" (true). It now gives a more helpful suggestion, which is what we can currently do to address this limitation with the resources we have.

Feel free to open a feature request for larger tables, if you think having more open issues is useful.

commit e5c12c8e (2022-08-25): kernel: hashlib: cleaned up message about table size in cases where `sizeof(int) == 4`, (closes #3440)
commit 6717e020 (2022-08-25): kernel: hashlib: cleaned up message about table size in cases where `sizeof(int) == 4`, (closes #3440)