YosysHQ/yosys #3442: How to identify bit-level dffsr ？

1396510211, 2022-08-09:
## How to identify bit-level dffsr ？

Consider the following example
```verilog
module TOP(
	input reg [4:0] out,
	input wire clk,
	input wire reset,
	input wire set,
	input wire [4:0] in,
);

	always@(posedge clk, negedge reset, negedge set) begin
		if(!reset) begin
			out[0] = 0;
			out[1] = 0;
		end
		else if (!set) begin 
			out[0] = 1;
			out[1] = 1;
			out[2] = 1;
		end
		else begin
			out[0] = in[0];
			out[1] = in[1];
			out[2] = in[2];
			out[3] = in[3];
		end
	end

endmodule

```
When I do the following ：
```shell
read -sv demo.v 
hierarchy -top TOP 
proc
opt_dff
flatten
```
## Expected behavior

The expected result is that there are two bits of dffsr(that is out[0] and out[1]) , one bit of adff, and one bit of dffe.

## Actual behavior
``` verilog
  attribute \src "demo.v:9.2-25.5"
  cell $dffsr $procdff$20
    parameter \CLR_POLARITY 1'1
    parameter \SET_POLARITY 1'1
    parameter \CLK_POLARITY 1'1
    parameter \WIDTH 4
    connect \CLR $auto$proc_dff.cc:124:gen_dffsr_complex$19
    connect \SET $auto$proc_dff.cc:117:gen_dffsr_complex$17
    connect \CLK \clk
    connect \Q \out [3:0]
    connect \D \in [3:0]
  end
```

nakengelhardt, 2022-08-12:
The script you have posted does not at any point map to fine-grain cells, so I am not sure why you would expect that outcome? Mapping to fine-grain would involve a flow that runs `techmap` at some point.

1396510211, 2022-08-12:
When I execute the techmap pass :
```bash attribute \src "test.v:9.51-25.5"
  cell $_DFFSR_PPP_ $auto$90
    connect \Q \out [2]
    connect \D \in [2]
    connect \R $auto$19 [2]
    connect \S $auto$17 [2]
    connect \C \clk
  end

  attribute \src "test.v:9.51-25.5"
  cell $_DFFSR_PPP_ $auto$89
    connect \Q \out [1]
    connect \D \in [1]
    connect \R $auto$19 [1]
    connect \S $auto$17 [1]
    connect \C \clk
  end

  attribute \src "test.v:9.51-25.5"
  cell $_DFFSR_PPP_ $auto$88
    connect \Q \out [0]
    connect \D \in [0]
    connect \R $auto$19 [0]
    connect \S $auto$17 [0]
    connect \C \clk
  end

 attribute \src "test.v:9.51-25.5"
  cell $_DFFSR_PPP_ $auto$91
    connect \Q \out [3]
    connect \D \in [3]
    connect \R $auto$19 [3]
    connect \S $auto$17 [3]
    connect \C \clk
  end
```
Is there any way to convert it to the following ?
[image]

nakengelhardt, 2022-08-12:
`show`?

1396510211, 2022-08-12:
I mean the result of the synthesis on the graph you can see is  two bits of dffsr(that is out[0] and out[1]) , one bit of adffe(out[2]), and one bit of dffe (out[3]),while the result of yosys is four bits of dffsr (_DFFSR_PPP_)  

how can i convert  the dffsr(out[2] & out[3]) to dffe or adff?

nakengelhardt, 2022-08-12:
If you want the results of synthesis, you need to run synthesis. Which architecture are you targeting?

1396510211, 2022-08-12:
When I execute the synth pass ,still   four bits of dffsr
```bash
 attribute \src "demo.v:9.2-25.5"
  cell $_DFFSR_PPN_ $auto$ff.cc:262:slice$126
    connect \C \clk
    connect \D \in [0]
    connect \Q \out [0]
    connect \R \reset
    connect \S $abc$207$auto$proc_dff.cc:117:gen_dffsr_complex$17[1]
  end

  attribute \src "demo.v:9.2-25.5"
  cell $_DFFSR_PPN_ $auto$ff.cc:262:slice$127
    connect \C \clk
    connect \D \in [1]
    connect \Q \out [1]
    connect \R \reset
    connect \S $abc$207$auto$proc_dff.cc:117:gen_dffsr_complex$17[1]
  end

  attribute \src "demo.v:9.2-25.5"
  cell $_DFFSR_PPP_ $auto$ff.cc:262:slice$128
    connect \C \clk
    connect \D \in [2]
    connect \Q \out [2]
    connect \R $abc$207$auto$proc_dff.cc:124:gen_dffsr_complex$19[2]
    connect \S $abc$207$auto$proc_dff.cc:117:gen_dffsr_complex$17[2]
  end

  attribute \src "demo.v:9.2-25.5"
  cell $_DFFSR_PPP_ $auto$ff.cc:262:slice$129
    connect \C \clk
    connect \D \in [3]
    connect \Q \out [3]
    connect \R $abc$207$auto$proc_dff.cc:124:gen_dffsr_complex$19[3]
    connect \S $abc$207$auto$proc_dff.cc:117:gen_dffsr_complex$17[3]
  end
```

nakengelhardt, 2022-08-12:
Ah, I see. I don't think we have an optimization for that currently.

1396510211, 2022-08-12:
So it's just a difference and not a problem, because it's completely correct in form  right?

nakengelhardt, 2022-09-05:
Sorry for late reply. As far as I can tell, yes, both do the same function, just with different representation.

1396510211, 2022-09-05:
Thank you for your reply, this is indeed a problem that has been bothering me for a long time, I would appreciate it if you could let me know after you solve it. ^_^

nakengelhardt, 2022-09-05:
Can you tell me why this is a problem for you?

Yosys mainly targets FPGA architectures, so there usually is no difference in area between the different flop types as there is only one hardware primitive available to map to. I can see that it would be slightly more area for ASIC flows, but we have many other QoR problems that I would think would weigh more than this edge case...