YosysHQ/yosys #3446: Post synth simulation

tsp6, 2022-08-17:
I am using wsl system. In the post synthesis simulation, when i try to use the Icarus Verilog simulator I got this error : 

tsp@DESKTOP-S7PFFHU:~/Projects$ iverilog -o example.vvp example_synth.v example_tb.v yosys/gatemate/cells_sim.v
yosys/gatemate/cells_sim.v: No such file or directory
example_synth.v:17: error: Unknown module type: CC_LUT3
example_synth.v:23: error: Unknown module type: CC_BUFG
example_synth.v:32: error: Unknown module type: CC_DFF
example_synth.v:44: error: Unknown module type: CC_DFF
example_synth.v:56: error: Unknown module type: CC_DFF
example_synth.v:68: error: Unknown module type: CC_DFF
example_synth.v:80: error: Unknown module type: CC_DFF
example_synth.v:92: error: Unknown module type: CC_DFF
example_synth.v:104: error: Unknown module type: CC_DFF
example_synth.v:116: error: Unknown module type: CC_DFF
example_synth.v:123: error: Unknown module type: CC_IBUF
example_synth.v:127: error: Unknown module type: CC_OBUF
example_synth.v:131: error: Unknown module type: CC_OBUF
example_synth.v:135: error: Unknown module type: CC_OBUF
example_synth.v:139: error: Unknown module type: CC_OBUF
example_synth.v:143: error: Unknown module type: CC_OBUF
example_synth.v:147: error: Unknown module type: CC_OBUF
example_synth.v:151: error: Unknown module type: CC_OBUF
example_synth.v:155: error: Unknown module type: CC_OBUF
example_synth.v:159: error: Unknown module type: CC_IBUF
21 error(s) during elaboration.
*** These modules were missing:
        CC_BUFG referenced 1 times.
        CC_DFF referenced 8 times.
        CC_IBUF referenced 2 times.
        CC_LUT3 referenced 1 times.
        CC_OBUF referenced 8 times.
***

**This is the path of my files and yosys installation:**

tsp@DESKTOP-S7PFFHU:~$ ls
Projects  cc-toolchain-linux  ghdl  ghdl-yosys-plugin  yosys  yosys.log
tsp@DESKTOP-S7PFFHU:~$ cd Projects/
tsp@DESKTOP-S7PFFHU:~/Projects$ ls
adder.v  example.ccf  example_synth.v  python.ipynb   yosys.log
c++.cpp  example.v    example_tb.v     synth_adder.v
tsp@DESKTOP-S7PFFHU:~/Projects$

gatecat, 2022-08-17:
> yosys/gatemate/cells_sim.v: No such file or directory

Your path to `cells_sim.v` isn't correct (I'm somewhat surprised Icarus doesn't just give up at that point, but hey ho).

I think you probably want `../yosys/techlibs/gatemate/cells_sim.v`

tsp6, 2022-08-17:
It worked. thnanks

I also had an issue after generating vcd file. In order to view the vcd file, I have already installed the packages for GTK wave wavefor viewer. Then when I use this command: 

tsp@DESKTOP-S7PFFHU:~/Projects$ vvp -N example.vvp
VCD info: dumpfile example_tb.vcd opened for output.
tsp@DESKTOP-S7PFFHU:~/Projects$ gtkwave example_tb.vcd
Could not initialize GTK!  Is DISPLAY env var/xhost set?

Usage: gtkwave [OPTION]... [DUMPFILE] [SAVEFILE] [RCFILE]

  -n, --nocli=DIRPATH        use file requester for dumpfile name
  -f, --dump=FILE            specify dumpfile name
  -F, --fastload             generate/use VCD recoder fastload files
  -o, --optimize             optimize VCD to FST
  -a, --save=FILE            specify savefile name
  -A, --autosavename         assume savefile is suffix modified dumpfile name
  -r, --rcfile=FILE          specify override .rcfile name
  -d, --defaultskip          if missing .rcfile, do not use useful defaults
  -D, --dualid=WHICH         specify multisession identifier
  -l, --logfile=FILE         specify simulation logfile name for time values
  -s, --start=TIME           specify start time for LXT2/VZT block skip
  -e, --end=TIME             specify end time for LXT2/VZT block skip
  -t, --stems=FILE           specify stems file for source code annotation
  -c, --cpu=NUMCPUS          specify number of CPUs for parallelizable ops
  -N, --nowm                 disable window manager for most windows
  -M, --nomenus              do not render menubar (for making applets)
  -S, --script=FILE          specify Tcl command script file for execution
  -T, --tcl_init=FILE        specify Tcl command script file to be loaded on startup
  -W, --wish                 enable Tcl command line on stdio
  -R, --repscript=FILE       specify timer-driven Tcl command script file
  -P, --repperiod=VALUE      specify repscript period in msec (default: 500)
  -X, --xid=XID              specify XID of window for GtkPlug to connect to
  -1, --rpcid=RPCID          specify RPCID of GConf session
  -2, --chdir=DIR            specify new current working directory
  -3, --restore              restore previous session
  -4, --rcvar                specify single rc variable values individually
  -5, --sstexclude           specify sst exclusion filter filename
  -I, --interactive          interactive VCD mode (filename is shared mem ID)
  -C, --comphier             use compressed hierarchy names (slower)
  -g, --giga                 use gigabyte mempacking when recoding (slower)
  -L, --legacy               use legacy VCD mode rather than the VCD recoder
  -v, --vcd                  use stdin as a VCD dumpfile
  -O, --output=FILE          specify filename for stdout/stderr redirect
  -z, --slider-zoom          enable horizontal slider stretch zoom
  -V, --version              display version banner then exit
  -h, --help                 display this help then exit
  -x, --exit                 exit after loading trace (for loader benchmarks)

VCD files and save files may be compressed with zip or gzip.
GHW files may be compressed with gzip or bzip2.
Other formats must remain uncompressed due to their non-linear access.
Note that DUMPFILE is optional if the --dump or --nocli options are specified.
SAVEFILE and RCFILE are always optional.

Report bugs to <bybell@rocketmail.com>.

can I get any help for this

gatecat, 2022-08-17:
This looks like a GTKWave/WSL issue (probably you need an X server or similar set up but I'm not sure how the standard WSL approach is there). Not something related to Yosys at this point.

closed 2022-08-17 by gatecat