YosysHQ/yosys #3451: Quastion: does the open source FPGA/ASIC flow support *Dynamically Sensitizable Paths*

jeras, 2022-08-18:
### Introduction

I plan to write a RISC-V CPU where a critical timing path will behave like described in [this article](https://people.cs.nctu.edu.tw/~luc/pub/static_timing_analysis_of_dynamically_sensitizable_paths.pdf). I have some experience with specifying timing constraints, but i have never discussed the described problem, I had some luck with Google finding this article quickly. I do not have much insight into STA internals, so my nomenclature is improvised.

### Description of planed CPU

I plan to write a CPU with a shared instruction/load/store bus, this would split the execution of instructions into 2 phases and 2 pipeline stages. A single adder would be shared between calculating the instruction address (PC increment, branch, jump) in `PH2` and ld/st address or ALU in `PH2`. A simple W1/R1 port register file used for GPR.

I will try to explain the execution of `add` as a relevant example.
- `PH1` instruction address on bus.
- `PH2` instruction OP on read data bus, at the end of the clock period stored into buffer. `rs1` address decoded (from RAM rdata) and used by GPR register file, asynchronous read data goes into instruction buffer/register. There is no load/store. Adder is used to increment instruction address, result is registered.
- `PH1` instruction address on bus (next instruction in pipeline). `rs2` decoded (from instruction buffer) and used by GPR register file, asynchronous read data goes into ALU to calculate the `add` instruction result going into data buffer/register.
- `PH2` writeback result from data buffer/register into GPR register file. ...
- ...

### Timing paths

There are two timing paths through the GPR register file with asynchronous read.
- `PH1` (RAM read data delay) + (GPR register file decoder)
- `PH2` (GPR register file decoder) + (carry propagation in adder)

### Question

Would STA calculate PH1/2 timing separately or would it see the longest path between registers long paths on both sides of register file: (RAM read data delay) + (GPR register file decoder) + (carry propagation in adder)

Ravenslofty, 2022-08-22:
Yosys does not know about timing constraints; the actual timing checks are performed by place-and-route tooling like nextpnr, or the tooling used by the open ASIC flow.

nextpnr doesn't support or attempt to infer false-path constraints, so it would see the longest path between registers. I don't know about the oither programs.

jeras, 2022-08-22:
Thanks, the OpenRoad/Lane flow is using https://github.com/The-OpenROAD-Project/OpenSTA. I will ask there.

When I write the planed CPU, I will also create some tests with various tools. I suspect this would be one of the longest paths.

I will close this issue now, I might reopen it later, if I find any related issues.

closed 2022-08-22 by jeras