YosysHQ/yosys #3455: Commit a04b025ab broke my design.

M3m3M4n, 2022-08-23:
Long story short, my design simply stopped working. Tracking down changes ended up at Yosys and this commit: a04b025ab. I'm not familiar with Yosys internals. It'd be nice if the devs can explain what memory_libmap pass does, I can provide the build logs but I want to know what to look for first.

Ravenslofty, 2022-08-23:
We're going to need your design in order to debug and fix the issue you're having.

`memory_libmap` is a replacement for `memory_bram` and handles mapping of memories described in Verilog to actual memory blocks in the FPGA.

rowanG077, 2022-08-24:
"simply stopped working" is also kind off vague. What exactly stopped working.

It synthesizes without errors and place&route fails? It doesn't synthesize? It does synthesize and place&route successfully but simply doesn't work when placed on a FPGA, as if nothing is programmed at all?

M3m3M4n, 2022-08-24:
> "simply stopped working" is also kind off vague. What exactly stopped working.
> 
> It synthesizes without errors and place&route fails? It doesn't synthesize? It does synthesize and place&route successfully but simply doesn't work when placed on a FPGA, as if nothing is programmed at all?

Yes, it does synthesize and nextpnr runs successfully but the output bitstream does not work on FPGA. Sorry for being vague, I was just so pissed off after 2 days of debugging with verilator and turned out it was a yosys update.

> We're going to need your design in order to debug and fix the issue you're having.
> 
> `memory_libmap` is a replacement for `memory_bram` and handles mapping of memories described in Verilog to actual memory blocks in the FPGA.

To provide some context, I'm building a risc-v core, no debugging module yet, so to debug only signal dump and GPIO for the time being. GPIO suddenly stopped working after updating to the mentioned commit, so it could be anything. The commit suggests block ram??? But in the log the libmap pass does touch gpio_out...

I cut out the relevant part in the log, included with the modules in attached file. 

Here is the log + module [mems.zip](https://github.com/YosysHQ/yosys/files/9413047/mems.zip)

chrbayer, 2022-09-09:
I have the same problem for ICE40, build a lot of yosys versions to find out that the corresponding ICE40 commit d7dc2313 break my design. I do not get an errors from yosys or later nextpnr, but the synthesized design just does not work. It is a 6502 deviate with bram banks for ROM and RAM. I noticed that for what ever reason these steps are removed from the synth command:  opt, wreduce, peepopt, opt_clean and s.

What additional information which could help should I provide?

nakengelhardt, 2022-09-09:
I guess the first step would be to narrow down if the issue is that the memories did not get mapped as expected by yosys, or if icebram or nextpnr is struggling with some of the new memory configuration options that `memory_libmap` can now use. Are you using icebram or is this broken with the original memory initialization contents from the RTL going through yosys+nextpnr? If you look at the output netlist of yosys, how does the instantiation of the bram modules look different? Are there any problems you might be able to see? Does adding `-no-rw-check` to `synth_ice40` change anything?

This design sounds reasonably small, could you share the code?

chrbayer, 2022-09-09:
I 'm indeed using icebram, and after eliminating it from the build process, it works again. So it seems that icebram is not compatible with the asc file from nextpnr. I will try to investigate the changes in the asc and try to fix icebram.