YosysHQ/yosys #3456: latch inferred when indexing with incremented integer

kranerup, 2022-08-23:
## Steps to reproduce the issue

[latch1.txt](https://github.com/YosysHQ/yosys/files/9402598/latch1.txt)

read_verilog -sv latch1.txt; proc

Yosys version used:
Yosys 0.9+4081 (git sha1 https://github.com/YosysHQ/yosys/commit/862e84eb3d85dfaacc7025b33c7dbe038bfff2d5, clang 10.0.0-4ubuntu1 -fPIC -Os)

## Expected behavior

There should be no latch inferred.

## Actual behavior

ERROR: Latch inferred for signal `\pa_top_switch_pb0.\dut_iPbu_iWrr_next_bval[1]' from always_comb process `\pa_top_switch_pb0.$proc$latch1.v:0$3'.

jix, 2022-08-23:
Like #3447, this is an issue with constant signals being recognized as such too late. When using `always @*` instead, the inferred latch has a constant active enable and will be removed by a later `opt_dff`. Improving tracking of constant assignments to variables during AST simplification would most likely fix both #3447 and this. AFAICT that would not be an easy task though.

Unlike #3447, this could also be fixed at a later stage by performing netlist level constant propagation within `proc`. Just for debugging I added `pmuxtree; opt_expr -keepdc` between `proc_mux` and `proc_dlatch` which makes this issue go away (but is untested and may cause all kind of other issues).

An alternative way to fix this would be to defer the always_comb check from `proc_dlatch` to the end of `proc`. This could be done using an attribute for latches indicating that they must end up with constant enables and a new sub-pass at the end of `proc` that verifies this and removes those latches.

As a workaround it is possible to remove the `always_comb` attribute from all processes between `read_verilog` and `proc` using `setattr -unset always_comb p:*` which omits the checks implied by `always_comb`.

Ravenslofty, 2022-08-23:
(as a side note, the version of Yosys you're using is a year old, and I would recommend upgrading)

kranerup, 2022-08-23:
The "setattr -unset always_comb p:*" workaround resolved the problem and a few other inferred latch issues. 

Yes, I'm aware it's an old version. I will try to upgrade.