YosysHQ/yosys #3464: Building tests broken on RISC-V since yosys 0.18

Popolon, 2022-08-26:
## Steps to reproduce the issue
Build the package on x86_64, it works, and with with the same version of packages, including Icarus Verilog, it fail.
The same libraries works on Arch Linux for x86_64, but fail on RISC-V with the following message:

```
[...]
Test: mem2reg_bounds_tern -> ok
Test: operators -> ERROR!
Note: Make sure that 'iverilog' is an up-to-date git checkout of Icarus Verilog.
make[1]: *** [../tools/autotest.mk:10: operators.v] Error 1
Test: macros -> ok
[...]
Test: case_large -> ok
make[1]: Leaving directory '/home/pop/yosys/src/yosys-yosys-0.20/tests/simple'
make: *** [Makefile:823: test] Error 2
```

*Provide instructions for reproducing the issue. Make sure to include
all necessary source files. (You can simply drag&drop a .zip file into
the issue editor.)*

build the package on Arch Linux for RISC-V by:
    make \
        PREFIX="/usr" \
        CONFIG=gcc \
        ENABLE_TCL=1 \
        ENABLE_ABC=1 \
        ENABLE_GLOB=1 \
        ENABLE_PLUGINS=1 \
        ENABLE_LIBYOSYS=1 \
        ENABLE_PROTOBUF=1 \
        ENABLE_ZLIB=1 \
        ENABLE_PYOSYS=1 \
        ABCEXTERNAL=abc \
        BOOST_PYTHON_LIB="-lpython3.10 -lboost_python310"
    make \
        PREFIX="/usr" \
        CONFIG=gcc \
        ENABLE_TCL=1 \
        ENABLE_ABC=1 \
        ENABLE_GLOB=1 \
        ENABLE_PLUGINS=1 \
        ENABLE_LIBYOSYS=1 \
        ENABLE_PROTOBUF=1 \
        ENABLE_ZLIB=1 \
        ENABLE_PYOSYS=1 \
        ABCEXTERNAL=abc \
        BOOST_PYTHON_LIB="-lpython3.10 -lboost_python310" test

or use makepkg with the last PKGBUILD and updating version number, sha512sum and URL (updated PKGBUILD at bottom

Also, make sure that the issue is actually reproducable in current git
master of Yosys.

See https://stackoverflow.com/help/mcve for some information on how to
create a Minimal, Complete, and Verifiable example (MCVE).

Please do not waste our time with issues that lack sufficient information
to reproduce the issue easily. We will simply close those issues.

Contact https://www.yosyshq.com/ if you need commercial support for Yosys.

## Expected behavior

*Please describe the behavior you would have expected from the tool.*

## Actual behavior

*Please describe how the behavior you see differs from the expected behavior.*

```
# Update: Popolon <popolon@popolon.org>
# Maintainer: Felix Yan <felixonmars@archlinux.org>
# Contributor: Benoît Allard <benoit dot allard at gmx dot de>
# Contributor: killruana <killruana@gmail.com>

pkgname=yosys
pkgver=0.20
pkgrel=1
pkgdesc='A framework for RTL synthesis'
arch=('x86_64' 'armv7h' 'aarch64' 'riscv32' 'riscv64')
url='http://yosyshq.net/yosys/'
license=('custom:ISC')
depends=('abc' 'bash' 'boost-libs' 'tcl' 'libffi' 'python' 'protobuf')
makedepends=('boost')
checkdepends=('iverilog')
optdepends=('graphviz: Schematics display support'
            'xdot: Display netlists')
source=("https://github.com/YosysHQ/yosys/archive/refs/tags/$pkgname-$pkgver.tar.gz")
sha512sums=('c1ae62e55928f26ff6142e9291cbd9b593b39f755d0294b4441bc66fa0eeec90bd71014495e54b5b9530ce609af8b007c755096295a0f7599c739535316c929b')

_make() {
    make \
        PREFIX="/usr" \
        CONFIG=gcc \
        ENABLE_TCL=1 \
        ENABLE_ABC=1 \
        ENABLE_GLOB=1 \
        ENABLE_PLUGINS=1 \
        ENABLE_LIBYOSYS=1 \
        ENABLE_PROTOBUF=1 \
        ENABLE_ZLIB=1 \
        ENABLE_PYOSYS=1 \
        ABCEXTERNAL=abc \
        BOOST_PYTHON_LIB="-lpython3.10 -lboost_python310" $@
}

build() {
    cd $pkgname-$pkgname-$pkgver
    _make
}

check() {
    cd $pkgname-$pkgname-$pkgver
    _make test
}

package() {
    cd $pkgname-$pkgname-$pkgver
    _make DESTDIR="$pkgdir" PYTHON_PREFIX="$pkgdir/usr" install

    install -Dm644 COPYING -t "$pkgdir"/usr/share/licenses/$pkgname/
}
```

mmicko, 2022-08-29:
Could you please pack yosys/tests/simple/operators.out  directory and yosys/tests/simple/operators.log file so we could take a look and compare with run on other platforms. I do not own any riscv64 physical system to test this, but maybe we can narrow down issue.

Popolon, 2022-08-30:
Thank you for your interest for this bug.

The directory are packed here (I put operators.err, has there was no operators.log and operators.out subdir) :
https://popolon.org/depots/ArchLinuxRISC-V/yosys/operators.tar.zst

Popolon, 2022-08-30:
I forgot to comment that the problem occur on both yosys 18,19, 20, that was not the case in yosys <18.

Popolon, 2022-10-11:
It looks likes the problem is solved, in archlinus package, there was a change in PKGBUILD in april I messed on 0.12:
https://github.com/archlinux/svntogit-community/commit/65d834ebd810ad85f9f0ae41fe1a51ba5cb58028

The package maintainer just upgraded to 0.20 last month (and ARM and RISC-V versions of Arch Linux have it too now) :).

Sadly, the URL is still the old one (that is a casino parking site), and old github source is used too. Need to catch attention of the package maintainer on these points.