YosysHQ/yosys #3476: 'Optimization level' support

ColonelPhantom, 2022-09-07:
### Feature Description

Yosys has many optimization-related features. However, they are not always enabled (because of synthesis time?), for example `synth_ice40` runs `opt`, but not e.g. `opt -full`, as far as I can see.

Would it be possible to implement something like optimization levels (e.g. `-O1` up to `-O(whatever)`) to be able to trade off between synthesis speed and final size? Or is this not realistic and/or are all (useful) optimization passes already enabled in most synth scripts?

Ravenslofty, 2022-09-10:
There were proposals for things like this, but it's difficult to classify things into tiers like this, because a lot of Yosys assumes other transformations have been made, and also optimisations are generally grouped together in `opt` or `proc` and iterated until done. If there's a `-O1` level, but `opt` still takes a long time refining the design, is it necessarily better?

pbreuer, 2023-02-18:
What kind of optimization does yosys do? I want it to minimize the final LC count. 
My _guess_ is that what yosys does by default produces wide and shallow (i.e. low latency) logic. Is there a way to bias it more towards narrow and deep, minimizing LC count?