YosysHQ/yosys #3479: Checksum changed for archived release 0.18-dev

j2kun, 2022-09-09:
### Version

0.18-dev

### Reproduction Steps

```
wget https://github.com/YosysHQ/yosys/archive/6b7efe12b79d4ed68af6a31f124213e101edbbb1.zip
shasum -a 256 6b7efe12b79d4ed68af6a31f124213e101edbbb1.zip
```

Some time between Friday, September 2, 2022 at 7:47:00 AM PDT and today at 7:54 AM PST, the sha256 checksum changed on this archived release. Is this an expected change due to some re-publishing action? I believe this release was at commit `6b7efe12b79d4ed68af6a31f124213e101edbbb1`

### Expected Behavior

Checksum is `6ab9e3fa8444ed5b350a371b68619a644ec9c53e908020a5d72c3dadea8b056d`

### Actual Behavior

Checksum is `40455c4fce124998da25d471d1de89587e1ed1a4c8d0cd5d5dab3880a2eb0664`

jix, 2022-09-09:
As far as I can tell, the files for archive URLs like this are generated on the fly by GitHub from the repository content and are not the result of any explicit publishing step. This means by default there is no guarantee of a stable checksum as the files within the archive could be reordered or compressed with different settings / zlib versions each time it is generated.

j2kun, 2022-09-09:
Fair enough. Should I expect a stable checksum from a tagged release?

jix, 2022-09-09:
I can't find any clear documentation on this from GitHub at all, but as they use the same URL schema, I would not count on them having a stable checksum either.

Xiretza, 2022-09-09:
I can say from experience (managing lots of AUR packages) that the release tarballs are very stable, I don't think I've ever had a checksum break without the upstream repo actually changing the tag.

j2kun, 2022-09-22:
I ran into this again today: we pinned in https://github.com/google/fully-homomorphic-encryption/commit/50a74343c3295bb7347557b98e6093b39cf4f10b to `0.20`, and as of September 22, 2022 at 1:00:30 AM PDT (when we noticed it), the sha256sum changed from `b81994853fb4a12b45afc6503568e84499025add8e51615c24b164cf2e8359a8` to `91088f6b2a8a417a0a1c29e1138822deeb2fc2b328fcdb5da057988d744ca886`.

As of `September 21, 2022 at 1:04:19 AM PDT` this was not the case.

Is there something I'm missing here? Did anything happen in the last day that would have caused the tag to change?

j2kun, 2022-10-11:
After speaking with GitHub support, it turns out that the only types of assets with guaranteed checksums are those under 

https://github.com/YosysHQ/yosys/releases/download/$tag/$asset

And in Yosys's case, it's only the `tar.gz` that is under that path, while the zip archives are auto-generated and subject to change whenever GitHub deploys some new software. Closing this issue. Thanks for the help!

closed 2022-10-11 by j2kun