YosysHQ/yosys #3485: ERROR: Assert `j < i' failed in kernel/mem.cc:495  -- dual ported RAM

GyrosGeier, 2022-09-19:
### Version

Yosys 0.19 (git sha1 2156e20)

### Reproduction Steps

Minimal repro seems to be

[registers.zip](https://github.com/YosysHQ/yosys/files/9595561/registers.zip)

    yosys> ghdl --std=08 registers.vhdl -e registers
    yosys> opt_mem

This is basically standard dual-ported RAM, and is inferred correctly from the VHDL.

### Expected Behavior

The tool should not crash when encountering dual-ported RAM.

### Actual Behavior

There is an assertion failure when checking the priority map -- both write ports are enabled, so there must be some priority ordering, but the failure happens whichever port gets priority (i.e. swapping the two `if` blocks for the write enables around does not help).

lethalbit, 2022-09-19:
Just adding that this also happens on attempting to `write_verilog`, however `write_json` and `write_rtlil` work.

Also happens as of `Yosys 0.21+10`

GyrosGeier, 2022-09-19:
Generated JSON for easier repro.

    > read_json registers.json
    > opt_mem

[registers.zip](https://github.com/YosysHQ/yosys/files/9602007/registers.zip)

lethalbit, 2022-09-26:
It turns this is a dup of #3364 and was a bug in GHDL https://github.com/ghdl/ghdl-yosys-plugin/commit/7aed75ba30157e52de737b2e082d2e1b6f82435d which as since been fixed.

closed 2022-09-26 by lethalbit