YosysHQ/yosys #3495: I having a issue with synth_ice40

wel97459, 2022-09-29:
### Version

Yosys 0.21+23 (git sha1 a9795c4fc, clang 14.0.0-1ubuntu1 -fPIC -Os)

### Reproduction Steps

Works:
`yosys -p "synth_ice40 -json hardware.json" -q top_one_modules.v`

Is not working:
`yosys -p "synth_ice40 -json hardware.json" -q top_two_modules.v`

Only thing changed in top_two_modules.v is the addition of the second module named LedGlow.

[top_module.zip](https://github.com/YosysHQ/yosys/files/9672687/top_module.zip)

### Expected Behavior

Running the same code just with the second module remove:
`yosys -p "synth_ice40 -json hardware.json" -q top_one_modules.v`

You get a working output that nextpnr-ice40 reads and outputs a working asc file.

### Actual Behavior

The json file is missing some parts and netnames.  And nextpnr-ice40 reports warnings:
```
nextpnr-ice40 --up5k --package sg48 --opt-timing --pcf-allow-unconstrained --json hardware.json --asc hardware.asc --pcf upduino.pcf
Warning: net 'clk_12Mhz' does not exist in design, ignoring clock constraint
Warning: IO 'io_led' is unconstrained in PCF and will be automatically placed
Warning: IO 'Core22_clk' is unconstrained in PCF and will be automatically placed
Warning: IO 'Core22_areaRst_newReset' is unconstrained in PCF and will be automatically place
```d

nakengelhardt, 2022-09-29:
(Have not tried to reproduce, but just a note: please always use the option `-top <top_module_name>` with the `synth_*` passes, otherwise you can be in situations where either no top module is designated at all, or yosys picks a random one. For your two module example, there is no guarantee which module gets picked as top level - it might just as well have chosen to only synthesize the `LedGlow` module.)

wel97459, 2022-09-29:
Yup, that's the issue. 
Thanks for your help.

closed 2022-09-29 by wel97459