YosysHQ/yosys #3498: failure to read Liberty

maliberty, 2022-10-03:
### Version

Yosys 0.13+15 (git sha1 bc027b2ca, gcc 8.3.1 -fPIC -Os)

### Reproduction Steps

Unpack the attached tgz and run "yosys < load.tcl".

### Expected Behavior

Both Liberty files are successfully parsed.

### Actual Behavior

yosys> read_liberty good.lib
1. Executing Liberty frontend.
Imported 1 cell types from liberty file.

yosys> read_liberty bad.lib

2. Executing Liberty frontend.
ERROR: Syntax error in liberty file on line 5.

maliberty, 2022-10-03:
[bug.tar.gz](https://github.com/YosysHQ/yosys/files/9700941/bug.tar.gz)

maliberty, 2022-10-03:
```
diff *lib
5c5
<       power_down_function : !a+b ; 
---
>       power_down_function : "!a+b" ; 
```
The quotes are not required by other proprietary tools and are not supplied in a commercial PDK.

lethalbit, 2022-10-11:
This is strange, as per the liberty syntax standard the `function` attributes are specified to be quoted strings (See Chapter 3 of the Liberty Reference Manual).

I'll tag this as discussion so we can talk about what to do at the next dev meeting

maliberty, 2022-10-11:
Synopsys often accepts things that are not legal per the manual and effectively forces all tools to do the same.

lethalbit, 2022-10-11:
Which is strange because they're the ones who specified the format in the first place. Either way, it'll be discussed at the next dev meeting what the plan of action is for this.

lethalbit, 2022-10-12:
Okay, talked with some other members of the team and it is indeed something we want to support, especially as other tools do so.

Therefore I've re-classified the issue so it's properly categorized.

closed 2023-05-21 by mmicko