YosysHQ/yosys #3505: Several tests fail

yurivict, 2022-10-09:
### Version

0.22

### Reproduction Steps

```make test```

### Expected Behavior

n/a

### Actual Behavior

```
Test: t_wide_read_a7r0w5b2 -> ok
Test: t_wide_write_a7r0w5b2 -> ok
gmake[2]: Leaving directory '/disk-samsung/freebsd-ports/cad/yosys/work/yosys-yosys-0.22/tests/memlib'
cd tests/bram && bash run-test.sh ""
generating tests..
PRNG seed: 542627
running tests..
gmake[2]: Entering directory '/disk-samsung/freebsd-ports/cad/yosys/work/yosys-yosys-0.22/tests/bram'
../../techlibs/common/simlib.v:1378: warning: Port 1 (A) of $bmux expects 1 bits, got 32.
../../techlibs/common/simlib.v:1378:        : Pruning (signed) 31 high bits of the expression.
Passed memory_bram test 00_01.
../../techlibs/common/simlib.v:1378: warning: Port 1 (A) of $bmux expects 1 bits, got 32.
../../techlibs/common/simlib.v:1378:        : Pruning (signed) 31 high bits of the expression.
Passed memory_bram test 00_02.
../../techlibs/common/simlib.v:1378: warning: Port 1 (A) of $bmux expects 1 bits, got 32.
../../techlibs/common/simlib.v:1378:        : Pruning (signed) 31 high bits of the expression.
Passed memory_bram test 00_03.
../../techlibs/common/simlib.v:1378: warning: Port 1 (A) of $bmux expects 1 bits, got 32.
../../techlibs/common/simlib.v:1378:        : Pruning (signed) 31 high bits of the expression.
Passed memory_bram test 00_04.
../../techlibs/common/simlib.v:1378: warning: Port 1 (A) of $bmux expects 1 bits, got 32.
../../techlibs/common/simlib.v:1378:        : Pruning (signed) 31 high bits of the expression.
Passed memory_bram test 01_00.
../../techlibs/common/simlib.v:1378: warning: Port 1 (A) of $bmux expects 1 bits, got 32.
../../techlibs/common/simlib.v:1378:        : Pruning (signed) 31 high bits of the expression.
Passed memory_bram test 01_02.
../../techlibs/common/simlib.v:1378: warning: Port 1 (A) of $bmux expects 1 bits, got 32.
../../techlibs/common/simlib.v:1378:        : Pruning (signed) 31 high bits of the expression.
Passed memory_bram test 01_03.
../../techlibs/common/simlib.v:1378: warning: Port 1 (A) of $bmux expects 1 bits, got 32.
../../techlibs/common/simlib.v:1378:        : Pruning (signed) 31 high bits of the expression.
Passed memory_bram test 01_04.
../../techlibs/common/simlib.v:1378: warning: Port 1 (A) of $bmux expects 1 bits, got 32.
../../techlibs/common/simlib.v:1378:        : Pruning (signed) 31 high bits of the expression.
Passed memory_bram test 02_00.
../../techlibs/common/simlib.v:1378: warning: Port 1 (A) of $bmux expects 1 bits, got 32.
../../techlibs/common/simlib.v:1378:        : Pruning (signed) 31 high bits of the expression.
Passed memory_bram test 02_01.
../../techlibs/common/simlib.v:1378: warning: Port 1 (A) of $bmux expects 1 bits, got 32.
../../techlibs/common/simlib.v:1378:        : Pruning (signed) 31 high bits of the expression.
Passed memory_bram test 02_03.
../../techlibs/common/simlib.v:1378: warning: Port 1 (A) of $bmux expects 1 bits, got 32.
../../techlibs/common/simlib.v:1378:        : Pruning (signed) 31 high bits of the expression.
Passed memory_bram test 02_04.
../../techlibs/common/simlib.v:1378: warning: Port 1 (A) of $bmux expects 1 bits, got 32.
../../techlibs/common/simlib.v:1378:        : Pruning (signed) 31 high bits of the expression.
Passed memory_bram test 03_00.
../../techlibs/common/simlib.v:1378: warning: Port 1 (A) of $bmux expects 1 bits, got 32.
../../techlibs/common/simlib.v:1378:        : Pruning (signed) 31 high bits of the expression.
Passed memory_bram test 03_01.
../../techlibs/common/simlib.v:1378: warning: Port 1 (A) of $bmux expects 1 bits, got 32.
../../techlibs/common/simlib.v:1378:        : Pruning (signed) 31 high bits of the expression.
Passed memory_bram test 03_02.
../../techlibs/common/simlib.v:1378: warning: Port 1 (A) of $bmux expects 1 bits, got 32.
../../techlibs/common/simlib.v:1378:        : Pruning (signed) 31 high bits of the expression.
[... 217 lines cut ...]
ERROR: Selection @foo is not defined!
Expected error pattern 'Selection @foo is not defined!' found !!!
Running warn_empty_select_arg.ys..
Warning: Selection "foo" did not match any module.
Warning: Selection "bar" did not match any object.
cd tests/sat && bash run-test.sh
gmake[2]: Entering directory '/disk-samsung/freebsd-ports/cad/yosys/work/yosys-yosys-0.22/tests/sat'
Passed asserts.ys
Passed asserts_seq.ys
Passed bug2595.ys
Warning: Complex async reset for dff `\q [12]'.
Warning: Complex async reset for dff `\q [8]'.
Passed clk2fflogic.ys
Passed counters-repeat.ys
Passed counters.ys
Passed dff.ys
Passed expose_dff.ys
ERROR: Error opening 'grom.fst' as FST file
gmake[2]: *** [run-test.mk:46: grom.ys] Error 1
gmake[2]: Leaving directory '/disk-samsung/freebsd-ports/cad/yosys/work/yosys-yosys-0.22/tests/sat'
gmake[1]: *** [Makefile:831: test] Error 2
gmake[1]: Leaving directory '/disk-samsung/freebsd-ports/cad/yosys/work/yosys-yosys-0.22'
*** Error code 1

```

lethalbit, 2022-10-11:
I suspect this to be a FreeBSD related issue, as it doesn't seem to happen on our CI, nor can I reproduce it locally on my own system.

I'll set up a FreeBSD VM to test in when I have a chance to see if I can reproduce it then.

lparkes, 2022-10-15:
FYI I see this same error on NetBSD when compiling with clang. (Compiling Yosys with GCC on NetBSD doesn't get this far through the tests).

lparkes, 2022-10-15:
I have zeroed in on the code in libs/fst/fstapi.cc:fstReaderInit() where it starts with "if (zhandle)". About line 4276.

So what appears to be happening on NetBSD is that the initial read of grom.fst via stdio is buffered and so even though ftell() reports that file position is 17, the seek position of the underlying file descriptor is 905 (the end of the file). The file descriptor is then dup()ed and the new file descriptor's seek position is shared with the original file descriptor and so it is still 905. 

gzread() then has nothing to read because we are at EOF. 

I added the line "lseek(zfd, ftell(xc->f), SEEK_SET);" just after line 4274 so that the new file descriptor was positioned where we  have asked stdio to read upto.

This unit test still doesn't work, but at least it doesn't exit after failing to read the FST file.

lparkes, 2022-10-16:
I think I've fixed this and I've created PR #3512 for it. It's just two lines of code.

mmicko, 2022-10-24:
Fixed with https://github.com/YosysHQ/yosys/pull/3512

closed 2022-10-24 by mmicko