YosysHQ/yosys #3506: Current build broken "No rule to make target 'new', needed by 'src/sat/glucose/AbcGlucose.o"

janschiefer, 2022-10-09:
### Version

Build of GIT version 0e13d7e4c791a261a8746fefbbe92f1f3c68852d

### Reproduction Steps

git rev-parse --verify HEAD
0e13d7e4c791a261a8746fefbbe92f1f3c68852d

make clean
make clang-config OR make config-gcc ( happens with both )
make -n

`From https://github.com/YosysHQ/abc
 * branch              HEAD       -> FETCH_HEAD
+ git checkout ab5b16e
HEAD is now at ab5b16ed Additional fix for large liberty files
rm -f abc/abc-[0-9a-f]*
make -C abc -s CC="clang" CXX="clang" ABC_USE_LIBSTDCXX=1 ABC_USE_NAMESPACE=abc VERBOSE=@ ARCHFLAGS="-DABC_USE_STDINT_H -Wno-c++11-narrowing """ PROG="abc-ab5b16e" MSG_PREFIX="[ 95%] ABC: " 
[ 95%] ABC: Using CC=clang
[ 95%] ABC: Using CXX=clang
[ 95%] ABC: Using AR=ar
[ 95%] ABC: Using LD=clang
[ 95%] ABC: Compiling in namespace 
[ 95%] ABC: Compiling with CUDD
[ 95%] ABC: Using libreadline
[ 95%] ABC: Using pthreads
[ 95%] ABC: Using explicit -lstdc++
[ 95%] ABC: Using CFLAGS=-Wall -Wno-unused-function -Wno-write-strings -Wno-sign-compare -DABC_USE_STDINT_H -Wno-c++11-narrowing  -DABC_NAMESPACE=abc -fpermissive -x c++ -DABC_USE_CUDD=1 -DABC_USE_READLINE  -DABC_USE_PTHREADS
**make[1]: *** No rule to make target 'new', needed by 'src/sat/glucose/AbcGlucose.o'.  Stop.**
make: *** [Makefile:786: abc/abc-ab5b16e] Error 2
`

### Expected Behavior

Build working.

### Actual Behavior

Build fails due to missing target "new" for ABC-Glucose solver.

lethalbit, 2022-10-11:
I can't seem to reproduce the issue, as well as the CI builds are passing, are you willing to try a fresh build to see if you still run into this problem?

For reference the exact steps I did to try to reproduce this are as follows:
```sh
$ git clone https://github.com/YosysHQ/yosys.git
$ cd yosys
$ git checkout 0e13d7e4c
$ make clean # Done just in case something strange in clean was causing it
$ make -n
```

And the build completed all the way through.

janschiefer, 2022-10-11:
Still the same.

Interestingly if i just run "make" the following happens:
`[Makefile.conf] CONFIG := clang
[  0%] Building kernel/version_6e907acf8.cc
[  0%] Building kernel/version_6e907acf8.o
[  0%] Building kernel/driver.o
In file included from kernel/driver.cc:20:
**./kernel/yosys.h:42:10: fatal error: 'map' file not found**
#include <map>
         ^~~~~
1 error generated.
make: *** [Makefile:737: kernel/driver.o] Fehler 1
`

It seems the build environment is broken.

Output of make -nd:
`(...)
No implicit rule found for 'Makefile'.
  Finished prerequisites of target file 'Makefile'.
 No need to remake target 'Makefile'.
Updating goal targets....
Considering target file 'all'.
 File 'all' does not exist.
  Considering target file 'abc-ab5b16e'.
   File 'abc-ab5b16e' does not exist.
    Considering target file 'src/sat/glucose/AbcGlucose.o'.
     File 'src/sat/glucose/AbcGlucose.o' does not exist.
     Looking for an implicit rule for 'src/sat/glucose/AbcGlucose.o'.
     Trying pattern rule with stem 'AbcGlucose'.
     Trying implicit prerequisite 'src/sat/glucose/AbcGlucose.c'.
     Trying pattern rule with stem 'AbcGlucose'.
     Trying implicit prerequisite 'src/sat/glucose/AbcGlucose.cc'.
     Trying pattern rule with stem 'AbcGlucose'.
     Trying implicit prerequisite 'src/sat/glucose/AbcGlucose.cpp'.
     Found an implicit rule for 'src/sat/glucose/AbcGlucose.o'.
      Considering target file 'src/sat/glucose/AbcGlucose.cpp'.
      File 'src/sat/glucose/AbcGlucose.cpp' was considered already.
      Pruning file 'src/sat/glucose/AbcGlucose.cpp'.
      Considering target file 'src/sat/glucose/System.h'.
      File 'src/sat/glucose/System.h' was considered already.
      Considering target file 'src/sat/glucose/IntTypes.h'.
      File 'src/sat/glucose/IntTypes.h' was considered already.
      Considering target file 'src/misc/util/abc_namespaces.h'.
      File 'src/misc/util/abc_namespaces.h' was considered already.
      Considering target file 'src/sat/glucose/ParseUtils.h'.
      File 'src/sat/glucose/ParseUtils.h' was considered already.
      Considering target file 'src/misc/zlib/zlib.h'.
      File 'src/misc/zlib/zlib.h' was considered already.
      Considering target file 'src/misc/util/abc_global.h'.
      File 'src/misc/util/abc_global.h' was considered already.
      Considering target file 'src/misc/zlib/zconf.h'.
      File 'src/misc/zlib/zconf.h' was considered already.
      Considering target file 'src/sat/glucose/Options.h'.
      File 'src/sat/glucose/Options.h' was considered already.
      Considering target file 'src/sat/glucose/Vec.h'.
      File 'src/sat/glucose/Vec.h' was considered already.
      **Considering target file 'new'.
      Recently tried and failed to update file 'new'.
make[1]: *** No rule to make target 'new', needed by 'src/sat/glucose/AbcGlucose.o'.  Stop.
Reaping losing child 0x55a284f37f70 PID 54815 
make: *** [Makefile:782: abc/abc-ab5b16e] Error 2
Removing child 0x55a284f37f70 PID 54815 from chain.**
`

janschiefer, 2022-10-11:
Well... the mystery continues.

After I purged all clang / clang-14 packages, "make config-gcc && make -j32" miraculously succeeds.

After re-installation of clang "make config-clang && make -j32" still fails with the following error:
[Makefile.conf] CONFIG := clang
[ 0%] Building kernel/version_6e907acf8.cc
[ 0%] Building kernel/version_6e907acf8.o
[ 0%] Building kernel/driver.o
In file included from kernel/driver.cc:20:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include
^~~~~
1 error generated.
make: *** [Makefile:737: kernel/driver.o] Fehler 1

This seems like a wired Ubuntu ( I am using Ubuntu 22.04.1 LTS) clang, C++ standard library and yosys / abc configuration issue...

However, I can build other projects with clang without problems ( most notably nextpnr )!!

This previous problem here seems related:
https://github.com/YosysHQ/yosys/issues/681

lethalbit, 2022-10-12:
If you add the following to the end of the `Makefile.conf`

```
CXX := clang++
CC := clang
```

does that end up resolving the issue?

janschiefer, 2022-10-12:
Nope.

make clang-config
(add stuff to Makefile.conf)
make -j32:

[Makefile.conf] CONFIG := clang
[Makefile.conf] CXX := clang++
[Makefile.conf] CC := clang
[  0%] Building kernel/driver.o
[  1%] Building kernel/register.o
[  1%] Building kernel/rtlil.o
[  1%] Building kernel/log.o
In file included from kernel/driver.cc:20:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
[  2%] Building kernel/calc.o
In file included from kernel/register.cc:20:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
[  2%] Building kernel/yosys.o
In file included from kernel/rtlil.cc:20:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
[  2%] Building kernel/binding.o
In file included from kernel/log.cc:20:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
[  3%] Building kernel/cellaigs.o
In file included from kernel/calc.cc:24:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
[  3%] Building kernel/celledges.o
In file included from kernel/yosys.cc:20:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
[  3%] Building kernel/satgen.o
In file included from kernel/binding.cc:20:
In file included from kernel/binding.h:23:
In file included from ./kernel/rtlil.h:20:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
In file included from kernel/cellaigs.cc:20:
In file included from ./kernel/cellaigs.h:23:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
[  4%] Building kernel/qcsat.o
[  4%] Building kernel/mem.o
In file included from kernel/celledges.cc:20:
In file included from ./kernel/celledges.h:23:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
[  4%] Building kernel/ffmerge.o
In file included from kernel/satgen.cc:20:
In file included from ./kernel/satgen.h:23:
In file included from ./kernel/rtlil.h:20:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
[  5%] Building kernel/ff.o
In file included from kernel/qcsat.cc:20:
In file included from ./kernel/qcsat.h:23:
In file included from ./kernel/satgen.h:23:
In file included from ./kernel/rtlil.h:20:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
In file included from In file included from kernel/ffmerge.cckernel/mem.cc::2020:
:
In file included from In file included from ./kernel/ffmerge.h./kernel/mem.h::2323:
:
In file included from ./kernel/ffinit.h./kernel/yosys.h:42:10:: 23:
./kernel/yosys.h:42:10: fatal error: fatal error: 'map' file not found
'map' file not found
#include <map>
         ^~~~~
#include <map>
         ^~~~~
[  5%] Building kernel/fstdata.o
In file included from kernel/ff.cc:20:
In file included from ./kernel/ff.h:23:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
1 error generated.
[  6%] Building libs/bigint/BigIntegerUtils.o
In file included from kernel/fstdata.cc:20:
In file included from ./kernel/fstdata.h:23:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
1 error generated.
1 error generated.
make: *** [Makefile:738: kernel/driver.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** [Makefile:738: kernel/binding.o] Error 1
make: *** [Makefile:738: kernel/log.o] Error 1
In file included from libs/bigint/BigIntegerUtils.cc:1:
libs/bigint/BigIntegerUtils.hh:5:10: fatal error: 'string' file not found
#include <string>
         ^~~~~~~~
1 error generated.
make: *** [Makefile:738: libs/bigint/BigIntegerUtils.o] Error 1
1 error generated.
make: *** [Makefile:738: kernel/register.o] Error 1
1 error generated.
make: *** [Makefile:738: kernel/cellaigs.o] Error 1
1 error generated.
make: *** [Makefile:738: kernel/calc.o] Error 1
1 error generated.
1 error generated.
make: *** [Makefile:738: kernel/ffmerge.o] Error 1
make: *** [Makefile:738: kernel/mem.o] Error 1
1 error generated.
make: *** [Makefile:738: kernel/yosys.o] Error 1
1 error generated.
1 error generated.
make: *** [Makefile:738: kernel/celledges.o] Error 1
1 error generated.
make: *** [Makefile:738: kernel/ff.o] Error 1
make: *** [Makefile:738: kernel/qcsat.o] Error 1
1 error generated.
make: *** [Makefile:738: kernel/fstdata.o] Error 1
1 error generated.
make: *** [Makefile:738: kernel/satgen.o] Error 1
1 error generated.
make: *** [Makefile:738: kernel/rtlil.o] Error 1

janschiefer, 2022-10-12:
**I actually fixed the build with clang.**

1. Adding the following to CXXFLAGS:
" -I/usr/include/c++/11 -I/usr/include/x86_64-linux-gnu/c++/11/ "

2. Adding the following to LDLIBS:
"-L/usr/lib/x86_64-linux-gnu/"

3. Create symlink:
sudo ln -s /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30 /usr/lib/x86_64-linux-gnu/libstdc++.so

So ultimately clang seems to have trouble finding the standard library headers and libraries in the particular yosys build config.(However, it works with other projects, such as nextpnr...)

janschiefer, 2022-10-13:
Another alternative is to use llvm's libc++ instead of libstdc++:

Change the following lines:
CXXFLAGS := $(CXXFLAGS) **-stdlib=libc++** -Wall -Wextra -ggdb -I. -I"$(YOSYS_SRC)" -MD -MP -D_YOSYS_ -fPIC -I$(PREFIX)/include
LDLIBS := $(LDLIBS) **-stdlib=libc++ -lc++** -lm

janschiefer, 2022-10-13:
Solution:

Apparently clang++ selected gcc-12 as the source of libstdc++ and not gcc-11, which is the standard library source om my system.

While I had libstdc++-11-dev installed, for some wired reason libstdc++-12-dev was not installed along with g++-12...
After installing the package libstdc++-12-dev on Ubuntu the build works without modifications.

closed 2022-10-13 by janschiefer

aolofsson, 2023-10-08:
Thank you for posting the solution! Just ran into the same issue on my system with ubuntu 22.04 and yosys main build.