YosysHQ/yosys #3529: Breaking down mem_v2 cells

flaviens, 2022-10-28:
### Version

Yosys 0.22+29 (git sha1 fc53a0a5c, gcc 11.2.0 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

[I created a repository here with steps explained](https://github.com/flaviens/memv2).

### Expected Behavior

No memory anymore, only standard cells / DFFs.

### Actual Behavior

```

   Number of wires:                 29
   Number of wire bits:            146
   Number of public wires:          17
   Number of public wire bits:      78
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                 13
     $mem_v2                         1
     $mux                           12
```

I may be doing something wrong here. The intent is to break down `mem_v2` using the `memory` pass or some other, but I cannot achieve it somehow. Is there a way to break this `mem_v2`?

nakengelhardt, 2022-11-14:
Dual-port memories with different clocks on the ports can't be implemented using FFs (FF cells can only be in one clock domain), so `memory_map` will leave them alone (it should print a warning message about this).

flaviens, 2022-11-14:
That makes sense, thank you for your answer!

closed 2022-11-14 by flaviens