YosysHQ/yosys #3541: assert topmod is not None

tjthejuggler, 2022-11-04:
### Version

Yosys 0.22+70 (git sha1 310281a96, clang  -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

I'm following along with this video:

https://youtu.be/VJsMLPGg4U4?t=1139

and so I have this code in hello.smt2:

module hello(input clk, input rst, output [3:0] cnt); 
reg [3:0] cnt = 0;
always @(posedge clk) begin 
    if (rst) 
        cnt <= 0; 
    else 
        cnt <= cnt + 1; 
end 

`ifdef FORMAL
    assume property (cnt != 10);
    assert property (cnt != 15);
`endif
endmodule

and I run 

$ yosys-smtbmc ~/projects/sva-tests/hello1.smt2

### Expected Behavior

STATUS: PASSED

### Actual Behavior

##   0:00:00  Solver: yices
Traceback (most recent call last):
  File "/usr/local/bin/yosys-smtbmc", line 451, in <module>
    assert topmod is not None
AssertionError

lethalbit, 2022-12-01:
Hey @tjthejuggler, It looks like you're trying to run verilog though smtbmc, you first need to transform the module using `yosys`.

Rather than putting your verilog in `hello1.smt2` put it in `hello1.v` and run the following command:

```
$ yosys -p 'read_verilog -formal hello1.v; prep -top hello -nordff; write_smt2 hello1.smt2'
```
Then you can run `yosys-smtbmc` on the `hello1.smt2` file and it should pass:

```
$ yosys-smtbmc hello1.smt2                                                                          
##   0:00:00  Solver: yices
##   0:00:00  Checking assertions in step 0..
##   0:00:00  Checking assertions in step 1..
##   0:00:00  Checking assertions in step 2..
##   0:00:00  Checking assertions in step 3..
##   0:00:00  Checking assertions in step 4..
##   0:00:00  Checking assertions in step 5..
##   0:00:00  Checking assertions in step 6..
##   0:00:00  Checking assertions in step 7..
##   0:00:00  Checking assertions in step 8..
##   0:00:00  Checking assertions in step 9..
##   0:00:00  Checking assertions in step 10..
##   0:00:00  Checking assertions in step 11..
##   0:00:00  Checking assertions in step 12..
##   0:00:00  Checking assertions in step 13..
##   0:00:00  Checking assertions in step 14..
##   0:00:00  Checking assertions in step 15..
##   0:00:00  Checking assertions in step 16..
##   0:00:00  Checking assertions in step 17..
##   0:00:00  Checking assertions in step 18..
##   0:00:00  Checking assertions in step 19..
##   0:00:00  Status: PASSED
```

nakengelhardt, 2022-12-01:
Only with the commercial [Tabby CAD Suite](https://www.yosyshq.com/tabby-cad-datasheet) version of yosys, as you need the verific frontend to parse SVA properties. I think you already have an eval license from us, right? Here is the manual for how to use SBY for formal verification of SVA properties: https://yosyshq.readthedocs.io/projects/sby/en/latest/quickstart.html

tjthejuggler, 2022-12-02:
@nakengelhardt Yes, I do already have an eval. Thanks so much for the information and link.

closed 2022-12-02 by tjthejuggler