YosysHQ/yosys #3545: Globally defined SV function breaks CXXRTL export

haflex, 2022-11-07:
### Minimal Working Example:
```systemverilog
//top.sv
function int f(input int p);
    f = p;
endfunction

module top(input int i, output int o);
mod m(
    .i(i),
    .o(o)
);
endmodule

(* cxxrtl_blackbox *)
module mod(input int i, output int o);
endmodule
```
```
read_verilog -sv top.sv
write_cxxrtl
```
### Expected result:
c++ code with `struct bb_p_mod` and `struct p_top`

### Actual Result:
c++ code with only `struct p_top`, which is empty

### Details:
Existence of the function f() ---although not used--- causes `write_cxxrtl` to produce incomplete output.
Removing `f()` or moving it into `top` or to the end of the file produces expect output.

arcturusannamalai, 2023-05-07:
problem exists in HEAD today

Verilog code looks re-generated and its OK
```Verilog
../yosys -qp 'read_verilog -sv top.sv;
> write_verilog;'
/* Generated by Yosys 0.28+12 (git sha1 aa3b5d2f8, clang 15.0.7 -fPIC -Os) */

(* cxxrtl_blackbox =  1  *)
(* cells_not_processed =  1  *)
(* src = "top.sv:13.1-14.10" *)
module mod(i, o
);
  (* src = "top.sv:13.22-13.23" *)
  input [31:0] i;
  wire [31:0] i;
  (* src = "top.sv:13.36-13.37" *)
  output [31:0] o;
  wire [31:0] o;
endmodule

(* cells_not_processed =  1  *)
(* src = "top.sv:5.1-10.10" *)
module top(i, o
);
  (* src = "top.sv:5.22-5.23" *)
  input [31:0] i;
  wire [31:0] i;
  (* src = "top.sv:5.36-5.37" *)
  output [31:0] o;
  wire [31:0] o;
  (* module_not_derived = 32'd1 *)
  (* src = "top.sv:6.5-9.2" *)
  mod m (
    .i(i),
    .o(o)
  );
endmodule
```

but the C++ code has skipped the blackbox interface,
```c++
#include <backends/cxxrtl/cxxrtl.h>

#if defined(CXXRTL_INCLUDE_CAPI_IMPL) || \
    defined(CXXRTL_INCLUDE_VCD_CAPI_IMPL)
#include <backends/cxxrtl/cxxrtl_capi.cc>
#endif

#if defined(CXXRTL_INCLUDE_VCD_CAPI_IMPL)
#include <backends/cxxrtl/cxxrtl_vcd_capi.cc>
#endif

using namespace cxxrtl_yosys;

namespace cxxrtl_design {

// \top: 1
// \src: top.sv:5.1-10.10
struct p_top : public module {
	// \src: top.sv:5.36-5.37
	/*output*/ value<32> p_o;
	// \src: top.sv:5.22-5.23
	/*input*/ value<32> p_i;
	// \hdlname: m o
	// \src: top.sv:6.5-9.2|top.sv:13.36-13.37
	/*outline*/ value<32> p_m_2e_o;
	p_top(interior) {}
	p_top() {
		reset();
	};

	void reset() override;
	bool eval() override;
	bool commit() override;

	void debug_eval();
	debug_outline debug_eval_outline { std::bind(&p_top::debug_eval, this) };

	void debug_info(debug_items &items, std::string path = "") override;
}; // struct p_top

void p_top::reset() {
}

bool p_top::eval() {
	bool converged = true;
	// \hdlname: m o
	// \src: top.sv:6.5-9.2|top.sv:13.36-13.37
	value<32> p_m_2e_o;
	// connection
	p_o = p_m_2e_o;
	return converged;
}

bool p_top::commit() {
	bool changed = false;
	return changed;
}

void p_top::debug_eval() {
}

CXXRTL_EXTREMELY_COLD
void p_top::debug_info(debug_items &items, std::string path) {
	assert(path.empty() || path[path.size() - 1] == ' ');
	items.add(path + "m i", debug_item(debug_alias(), p_i, 0));
	items.add(path + "m o", debug_item(debug_eval_outline, p_m_2e_o, 0));
	items.add(path + "o", debug_item(debug_eval_outline, p_m_2e_o, 0));
	items.add(path + "i", debug_item(p_i, 0, debug_item::INPUT|debug_item::UNDRIVEN));
}

} // namespace cxxrtl_design

extern "C"
cxxrtl_toplevel cxxrtl_design_create() {
	return new _cxxrtl_toplevel { std::unique_ptr<cxxrtl_design::p_top>(new cxxrtl_design::p_top) };
}
```

arcturusannamalai, 2023-05-07:
Fix is to use the *write_cxxrtl -nohierarchy* since the hierarchy pass is removing the unused blackbox.

(simply running the write_cxxrtl shows the output indicating blackbox is removed)
```text
2. Executing CXXRTL backend.

2.1. Executing HIERARCHY pass (managing design hierarchy).

2.1.1. Finding top of design hierarchy..
root of   0 design levels: mod                 
root of   1 design levels: top                 
Automatically selected top as design top module.

2.1.2. Analyzing design hierarchy..
Top module:  \top
Used module:     \mod

2.1.3. Analyzing design hierarchy..
Top module:  \top
Used module:     \mod
Removed 0 unused modules.
Removed 0 unused modules.

2.2. Executing FLATTEN pass (flatten design).
Deleting now unused module mod.
```

Updated run with *write_cxxrtl -nohierarchy* shows better expected code,
```c++
#if defined(CXXRTL_INCLUDE_CAPI_IMPL) || \
    defined(CXXRTL_INCLUDE_VCD_CAPI_IMPL)
#include <backends/cxxrtl/cxxrtl_capi.cc>
#endif

#if defined(CXXRTL_INCLUDE_VCD_CAPI_IMPL)
#include <backends/cxxrtl/cxxrtl_vcd_capi.cc>
#endif

using namespace cxxrtl_yosys;

namespace cxxrtl_design {

// \cxxrtl_blackbox: 1
// \cells_not_processed: 1
// \src: top.sv:13.1-14.10
struct bb_p_mod : public module {
	// \src: top.sv:13.36-13.37
	/*output*/ wire<32> p_o;
	// \src: top.sv:13.22-13.23
	/*input*/ value<32> p_i;

	void reset() override {
	}

	bool eval() override {
		bool converged = true;
		return converged;
	}

	bool commit() override {
		bool changed = false;
		if (p_o.commit()) changed = true;
		return changed;
	}

	void debug_info(debug_items &items, std::string path = "") override {
		assert(path.empty() || path[path.size() - 1] == ' ');
		items.add(path + "o", debug_item(p_o, 0, debug_item::OUTPUT|debug_item::DRIVEN_SYNC|debug_item::DRIVEN_COMB));
		items.add(path + "i", debug_item(p_i, 0, debug_item::INPUT|debug_item::UNDRIVEN));
	}

	static std::unique_ptr<bb_p_mod> create(std::string name, metadata_map parameters, metadata_map attributes);
}; // struct bb_p_mod

// \cells_not_processed: 1
// \src: top.sv:5.1-10.10
struct p_top : public module {
	// \src: top.sv:5.36-5.37
	/*output*/ wire<32> p_o;
	// \src: top.sv:5.22-5.23
	/*input*/ value<32> p_i;
	// \hdlname: m o
	// \src: top.sv:6.5-9.2|top.sv:13.36-13.37
	/*outline*/ value<32> p_m_2e_o;
	p_top(interior) {}
	p_top() {
		reset();
	};

	void reset() override;
	bool eval() override;
	bool commit() override;

	void debug_eval();
	debug_outline debug_eval_outline { std::bind(&p_top::debug_eval, this) };

	void debug_info(debug_items &items, std::string path = "") override;
}; // struct p_top

void p_top::reset() {
}

bool p_top::eval() {
	bool converged = false;
	// \hdlname: m o
	// \src: top.sv:6.5-9.2|top.sv:13.36-13.37
	value<32> p_m_2e_o;
	// connection
	p_o.next = p_m_2e_o;
	return converged;
}

bool p_top::commit() {
	bool changed = false;
	if (p_o.commit()) changed = true;
	return changed;
}

void p_top::debug_eval() {
}

CXXRTL_EXTREMELY_COLD
void p_top::debug_info(debug_items &items, std::string path) {
	assert(path.empty() || path[path.size() - 1] == ' ');
	items.add(path + "m i", debug_item(debug_alias(), p_i, 0));
	items.add(path + "m o", debug_item(debug_eval_outline, p_m_2e_o, 0));
	items.add(path + "o", debug_item(debug_eval_outline, p_m_2e_o, 0));
	items.add(path + "i", debug_item(p_i, 0, debug_item::INPUT|debug_item::UNDRIVEN));
}

} // namespace cxxrtl_design

```