YosysHQ/yosys #3549: CXXRTL edge detector not working for non-input clocks

haflex, 2022-11-10:
### Version

Yosys 0.22 (git sha1 f109fa3d4c5, gcc 11.3.0 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

```verilog
module top (input logic clk);
    logic[7:0] counter;
    logic clkdiv;
    always_ff @(posedge clk) clkdiv <= ~clkdiv;
    always_ff @(posedge clkdiv) counter <= counter + 1;
endmodule
```
export to c++ with `write_cxxrtl`, simulate with trivial testbench toggleing `clk` every cycle.

### Expected Behavior

`clkdiv` toggles at half the rate of `clk`.
`counter` increases at every `posedge` of `clkdiv`

### Actual Behavior

`counter` does not increase at all.
It seems this is related to the clock detection `posedge_p_clkdiv`.
It is called in `p_top::eval` at the very beginning, before a new value for `p_clkdiv.next` has been calculated.
By the next time `p_top::eval` is called `p_clkdiv` is already `commit`ed, thus missing the posedge.

whitequark, 2022-11-10:
Known bug. I have a CXXRTL redesign in the works that will address this.

haflex, 2022-11-18:
I won't create another bug report because I think it might be related, because it also deals with `eval()`-`commit()` interference.
I have this simple sv code:
```verilog
module top (
    input logic clk,
    output logic [15:0] out
);

logic [32:0] shift;
initial  shift = 'h14d5ba65;
always_ff @( posedge clk ) shift <= {shift[31:0], shift[32] ^ shift[13]};

assign out = shift[15:0];

endmodule
```
The `out` signal receives it's new value in the step after the `posedge clk`, while it should receive the value in the same step. The assign `p_out = p_shift.curr...` can only provide the correct value after the new `p_shift` value has been `commit()`ed.
However, `eval()` always returns `converged = true` thus preventing another delta cycle.
To be honest I do not exactly understand the semantics behind `converged`, but it seems odd that combinatorical logic is unable to settle here.

arcturusannamalai, 2023-05-07:
seems similar to https://github.com/YosysHQ/yosys/issues/3611