YosysHQ/yosys #3557: ABC: + rewrite failure when running yosys-abc

ramrajrl, 2022-11-22:
### Version

0.19

### On which OS did this happen?

Linux

### Reproduction Steps

71.1.1. Executing ABC.
Running ABC command: "/build/bin/yosys-abc" -s -f /tmp/yosys-abc-tRZss0/abc.script 2>&1

ABC: + refactor 
ABC: An error occurred during computation. The original network is restored.
ABC: + balance 
ABC: + rewrite 
ABC: yosys-abc: src/base/abc/abcAig.c:167: void abc::Abc_AigFree(abc::Abc_Aig_t*): Assertion `Vec_PtrSize( pMan->vStackReplaceOld ) == 0' failed.
ABC: Flags A, B, or C are not zero.
ERROR: ABC: execution of command ""/build/bin/yosys-abc" -s -f /tmp/yosys-abc-tRZss0/abc.script 2>&1" failed: return code 134.

Any ideas on how I can debug this? I have checked the /tmp folder and can see failure at ABC: + rewrite

### Expected Behavior

ABC: + mfs 
ABC: + strash 
ABC: + refactor 
ABC: + balance 
ABC: + rewrite 
ABC: + refactor 
ABC: + balance 
ABC: + rewrite 
ABC: + rewrite -z 
ABC: + balance 
ABC: + refactor -z 
ABC: + rewrite -z 
ABC: + balance 
.
In a passing instance of RTL synthesis.

### Actual Behavior

ABC: + rewrite step fails, RTL synthesis on a FPGA flow passes. The RTL code is sensitive information and uploading the testcase in an open platform is an issue. Could you please suggest how to go about debugging this failure?

nakengelhardt, 2022-11-25:
This isn't any abc script included in yosys, is it? Do you have any reason to think the problem is with yosys and not a bug in abc?

ramrajrl, 2022-11-25:
As the link suggests, it perhaps has to do with Technology mapping part of it, but this is openlane-sky130 platform. https://github.com/YosysHQ/yosys/blob/8e640663d6b7ff84043068f48ed5f3cf7bff4321/passes/techmap/abc9_exe.cc 
Will also add that only some RTL configurations hit these issues, there isn't any structural change to RTL components or code. A comparable example would be different memory size instances, they are using the same underlying blocks even if the sizes are different. Thanks for your inputs.

jix, 2022-11-28:
There were issues with abc's rewrite command before (see #174) and the recommendation by abc's author is to use newer alternative commands since there are no plans to address the issue with abc's rewrite command. Yosys itself doesn't call abc's rewrite command in any of the abc scripts it uses, so this is not a Yosys issue.

closed 2022-11-28 by jix

ramrajrl, 2022-11-28:
Thank you for the insight, this is exactly the problem. By simply using the newer `drw` command instead of `rewrite` as suggested above fixes the ABC failure.