YosysHQ/yosys #3559: Documentation: `memory_libmap` requires an argument

xobs, 2022-11-25:
### Version

Yosys 0.23+21 (git sha1 e56c68996, gcc 11.2.0-19ubuntu1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Create a RAM with the following:

```
ram block $__SKY130_1K_ {
        abits 8;
        widths 32;
}
```

Then import it with:

```
yosys -p 'memory_libmap rams.txt'
```

### Expected Behavior

According to the documentation at https://github.com/YosysHQ/yosys/blob/982a11c709b4b363f85ae52a127f8a98bda30a3f/passes/memory/memlib.md#ram-dimensions this is a valid RAM.

### Actual Behavior

Yosys appears to want `global` or `per_port`:

```
1. Executing MEMORY_LIBMAP pass (mapping memories to cells).
ERROR: /opt/Si/src/OpenLane/scripts/yosys/sky130rams.txt:3: expected `global`, or `per_port`, got `;`.
```

Omitting this value, which is what is described in the documentation, appears to not be legal.

mmicko, 2022-11-25:
Note that you have typo, it should be `width` instead of `widths`.  But after that you would get error:
ERROR: rams.txt:1: at least one port group should be specified.
So you need to define read/write port group as well.

xobs, 2022-11-25:
Subtle. My mistake. Thank you for clarifying. I was adapting an existing design, and didn't notice the difference in pluralization.

closed 2022-11-25 by xobs