YosysHQ/yosys #3564: BLIF outputed by yosys involves DFF, and the BLIF file cannot be read by ABC

lesliepy99, 2022-11-28:
### Version

yosys --0.23

### On which OS did this happen?

Linux

### Reproduction Steps

I'm new to yosys and ABC for logic synthesis. I downloaded one design [aes_core](https://opencores.org/projects/apbtoaes128) from opencores, and run the following script with yosys to map the design into blif:
```
read_verilog ./designs/apbtoaes128/trunk/rtl/*.v
hierarchy -check -top aes_core
proc
techmap -map ./oss-cad-suite/share/yosys/adff2dff.v
synth
dfflibmap -prepare ./yosys-yosys-0.23/manual/PRESENTATION_Intro/mycells.lib 
abc -liberty ./yosys-yosys-0.23/manual/PRESENTATION_Intro/mycells.lib
dfflibmap -liberty ./yosys-yosys-0.23/manual/PRESENTATION_Intro/mycells.lib 
write_blif  -gates ./designs/aes_core.blif
``` 
After this, the blif only contains five types of gates (BUF, NOT, NAND, NOR, DFF); one snippet of the blif file is as follows:

```
...
.gate DFF C=clk D=$auto$rtlil.cc:2560:MuxGate$25762 Q=rd_count[0]
.gate DFF C=clk D=$auto$rtlil.cc:2560:MuxGate$25766 Q=rd_count[1]
.gate DFF C=clk D=$auto$rtlil.cc:2560:MuxGate$25770 Q=rd_count[2]
.gate DFF C=clk D=$auto$rtlil.cc:2560:MuxGate$25774 Q=rd_count[3]
.gate DFF C=clk D=$abc$11428$auto$fsm_map.cc:170:map_fsm$2040[0] Q=state[0]
.gate DFF C=clk D=$abc$11428$auto$fsm_map.cc:170:map_fsm$2040[1] Q=state[1]
.gate DFF C=clk D=$abc$11428$auto$fsm_map.cc:170:map_fsm$2040[2] Q=state[2]
.gate DFF C=clk D=$abc$11428$auto$fsm_map.cc:118:implement_pattern_cache$2077 Q=state[3]
.gate DFF C=clk D=$abc$11428$auto$fsm_map.cc:170:map_fsm$2040[4] Q=state[4]
...
```
At last, I wish to use ABC to read the blif file, the script I used with ABC is:
```
read ./yosys-yosys-0.23/manual/PRESENTATION_Intro/mycells.lib 
read_blif ./designs/aes_core.blif
```

### Expected Behavior

The BLIF is read by ABC.

### Actual Behavior

The output is:
```
Generic file reader requires a known file extension to open "./yosys-yosys-0.23/manual/PRESENTATION_Intro/mycells.h".
Line 393: Cannot find gate "DFF" in the library.
Reading network from file has failed.
```
It seems that when I read the cell library in ABC, the sequential gate is skipped, and I wonder the reason of this and how can we fix this issue.

Ravenslofty, 2022-12-01:
I suspect that the answer is that ABC does not support loading sequential cells from cell libraries; e.g. berkeley-abc/abc#121 is a similar issue where sequential flops in GENLIB format are also ignored. ABC does have the concept of flops, and Yosys can pass those through to ABC via `abc -dff`, but in this case the cells have already been mapped; when passing a netlist with flop cells to ABC in the `abc` pass, Yosys turns the outputs of flops into primary inputs, and the inputs of flops into primary outputs, thus hiding the flops from ABC, and I think failing to load netlists with flops in them is why Yosys does this.

lesliepy99, 2022-12-04:
> I suspect that the answer is that ABC does not support loading sequential cells from cell libraries; e.g. [berkeley-abc/abc#121](https://github.com/berkeley-abc/abc/issues/121) is a similar issue [...]

lesliepy99, 2022-12-04:
> I suspect that the answer is that ABC does not support loading sequential cells from cell libraries; e.g. [berkeley-abc/abc#121](https://github.com/berkeley-abc/abc/issues/121) is a similar issue [...]

Understood, I just find another script for the similar purpose, thanks a lot!

closed 2022-12-04 by lesliepy99