YosysHQ/yosys #3571: Blowup in number of cells when swapping an assign and an always block

dwRchyngqxs, 2022-12-01:
### Version

Yosys 0.23+38 (git sha1 f9db7c059, clang 10.0.0-4ubuntu1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Copy the below codes into two files, synthesize them targeting the ecp5 FPGA (I used the command `yosys -p "read_verilog <input_file>; synth_ecp5; write_verilog <output_file>`) and note the report produced by `stat`.  
Code 1:
```verilog
module top(y, clk, a, b, c, d, e);
  output wire y;
  input wire clk;
  input wire signed [18:0] a;
  input wire [13:0] b;
  input wire signed [15:0] c;
  input wire [8:0] d;
  input wire [10:0] e;
  reg r = 0;

  assign y = b[3:2] ? r : 0;
  always @(posedge clk)
    r <= d ? b & ((b ? a : c) && {d, b})
      : (a > ((e ? b : 1) ? e
        : d));
endmodule
```
Code 2:
```verilog
module top(y, clk, a, b, c, d, e);
  output wire y;
  input wire clk;
  input wire signed [18:0] a;
  input wire [13:0] b;
  input wire signed [15:0] c;
  input wire [8:0] d;
  input wire [10:0] e;
  reg r = 0;

  always @(posedge clk)
    r <= d ? b & ((b ? a : c) && {d, b})
      : (a > ((e ? b : 1) ? e
        : d));
  assign y = b[3:2] ? r : 0;
endmodule
```
The codes above are the same except the assign block swapped its position with the always block.  
The `stat` reports below show a factor 4 between the number of cells.
Report for code 1:
```
=== top ===

   Number of wires:                218
   Number of wire bits:            341
   Number of public wires:         218
   Number of public wire bits:     341
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                238
     CCU2C                           9
     L6MUX21                        43
     LUT4                          126
     PFUMX                          59
     TRELLIS_FF                      1
```
Report for code 2:
```
=== top ===

   Number of wires:                 36
   Number of wire bits:            169
   Number of public wires:          36
   Number of public wire bits:     169
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                 57
     CCU2C                           9
     L6MUX21                         2
     LUT4                           38
     PFUMX                           7
     TRELLIS_FF                      1
```
I tried to reduce the examples as much as I could (expression size, signal bit width) but the effect disappears. The only exception is the indexing on `b`, changes on it don't always make the blowup in number of cells disappears.

### Expected Behavior

The difference between reported number of cells should be small or nonexistent.

### Actual Behavior

The ratio between the reported number of cells is 4 (57 vs. 238).

gatecat, 2022-12-01:
With synthesis being so heuristic based it's hard to altogether avoid these cases, although there's always room for optimiser improvements. In this case "synth_ecp5 -abc9", which uses some newer ABC9 functionality for better optimisation, gives consistent results for both permutations of:
```
   Number of cells:                 68
     CCU2C                           9
     L6MUX21                         6
     LUT4                           44
     PFUMX                           8
     TRELLIS_FF                      1
```
If you want to optimise more aggressively for area you can use "-nowidelut -abc9" to avoid mapping wide LUT+MUX2 cascades:

```
   Number of cells:                 42
     CCU2C                           9
     LUT4                           32
     TRELLIS_FF                      1
```

dwRchyngqxs, 2022-12-01:
Thank you for the fast answer.

closed 2022-12-01 by dwRchyngqxs