YosysHQ/yosys #3574: Yosys 0.23+45 (git sha1 a64ed824e, clang 10.0.0-4ubuntu1 -fPIC -Os)

ratul619, 2022-12-04:
### Feature Description

Hi , 

How to set dont use library cells for some library cells based on pattern : like set_dont_use */XOR3* 

Using version : Yosys 0.23+45 (git sha1 a64ed824e, clang 10.0.0-4ubuntu1 -fPIC -Os)

Thanks

nakengelhardt, 2022-12-04:
Please provide more context. There are many ways to map to many different kinds of cell libraries with yosys. Without more details, the only general advice we can give is that if you don't want to map to these cells, don't include them in the library when asking to map.

ratul619, 2022-12-04:
hi , not different kind of cell libraries. For the same library , how can we set dont use for certain library cells ? so that they are not used in synthesis.  One way is to have dont_use true attribute in the library itself. Is there any option through command line ?

rbarzic, 2023-08-16:
This couldn't be supported by yosys directly because of the use of ABC  to do the mapping  - see #3638 - But ABC has a new option now to support that - so there is hope that Yosys could implement a set_dont_use command