YosysHQ/yosys #3594: Print backtrace on abort/assert

nakengelhardt, 2022-12-19:
### Feature Description

Currently it only prints the line where the assert was hit - it might help for some bug reports if it also prints more information about the context.

arcturusannamalai, 2023-05-07:
**libunwind** can be adapted do this in debug build of Yosys; however since Yosys is a command-by-command interpreter this is sufficient isn't it ? Unlike having function calls etc. its a straight line interpreter and it just fails on most recent command so I don't see a major need for it yet.

arcturusannamalai, 2023-05-09:
rough cut implementation to print a stack of commands executed to point of failure https://github.com/YosysHQ/yosys/pull/3757

arcturusannamalai, 2023-05-10:
Updated to actually print the stack trace; see commit detail

arcturusannamalai, 2023-07-01:
@nakengelhardt @whitequark - any idea on the PR ?