YosysHQ/yosys #3601: memory use reported is wrong

pbreuer, 2022-12-28:
### Version

Yosys 0.23 (git sha1 7ce5011c24b)

### On which OS did this happen?

Linux

### Reproduction Steps

Run yosys any way you like, but so it uses more than RAM capacity at peak, needing swap.

### Expected Behavior

The final MEM: output should be the peak memory footprint required.

### Actual Behavior

What is actually reported is the peak memory footprint in RAM, it doe not include swap space needed.

E.g. _End of script. Logfile hash: adee604607, CPU: user 398333.78s system 10906.60s,
**MEM: 89465.26 MB** peak_

That is false. The RAM on that machine is 90GB and the process used it all, maxing at 153GB RAM+swap through about 110h of cpu time, at between 50 and 100% cpu. I watched it throughout with htop. An earlier try with only 90GB+0.5GB swap enabled failed with kernel out of memory (OOM) reaper having it killed.

The problem is getrusage() in ./kernel/driver.cc, which gets the process data into ru_buffer. It's not your fault, but getrusage doesn't record peak _memory_ use (the ru_time field) as the man page for getrusage vaguely claims, it records the RAM usage.

The only fix is to get the memory usage the way htop and other similar utilities do it ... which I am not going to look up but which i would guess is probably via some function in a ps-misc library that polls data in /proc.

The reason one needs the right info out is so that I _don't_ have to watch the process myself through 110h of cpu time in order to know how much swap I really can get away with next time round.

pbreuer, 2022-12-28:
I've had a look and instead of getrusage you want to read field no 23 of /proc/the_pid/stat. See proc(5):

(23) vsize  %lu
                     Virtual memory size in bytes.

Alternatively you could parse /proc/the_pid/status for a line like

VmSize:  1900148 kB

According to proc(5), this is the same data as field no 1 of /proc/the_pid/statm:

    size       (1) total program size
                             (same as VmSize in /proc/pid/status)

I'd maybe go for reading stat or statm for the  resident size including swap, as one can reliably read it with scanf %lu, the manual says.

jix, 2023-01-02:
The current behavior is the best yosys can reasonably do. While using `/proc` you can query the _current_ size of virtual memory, you only get the _peak_ value for the resident set size. And while the resident set size doesn't count swap, the size of virtual memory is often larger than the actual amount of RAM used, sometimes by a lot. 

Measurements beyond what's easily available via getrusage are best left to dedicated tools. AFAIK, to get accurate measurements on Linux (including the peak amount of swap used), it's necessary to run processes in a separate control group and use the memory resource controller (see https://docs.kernel.org/admin-guide/cgroup-v2.html#memory). That is certainly out of scope for yosys though.

closed 2023-01-02 by jix

pbreuer, 2023-01-02:
'the best yosys can reasonably do' - absolute claims are  fraught with risk ... One CAN measure the virtual memory used after each pass, via /proc as inidcated (if available, else don't bother) taking the max with what was known already. That's good enough for practical purposes (*).

 Say the new vmem measure is approximate, but please do NOT claim the actual measure is the peak "memory" used, because it is not. 

Add the word "resident" before the actual measure and it will be right, just not useful exactly when one needs it to be (when one doesn't have enough RAM for the whole thing at one shot and needs to know how much would have been needed).

(*) I have seen no indication that any memory is released at the end of each pass, but perhaps some may go. Less than 1GB in 150GB  from what I am seeing. 

Also I guess that each pass writes the output into a space separate from the input space, and the next pass reads the old output , or maybe the old output is overwrtten into the input space. Anyway, I don't see the old input being freed and I would if it were ... which is the only scenario in which one would get a false low reading by measuring vmem at the end of each pass.

So I am pretty certain reading vmem at the end of each pass does the job. No polling required.

Also ... have you thought about the buffer/cache thrashing behavior implied by what seems to be reading serially through input, then writing separate output, and not freeing input? You would force the kernel to write the input to swap, very slowly, while in-kernel buffers take the output flow. You're OK if you now flip references
to input and output space and start reading what you just wrote for the next pass ... that comes from buffers. But if you write old output to input space you will be evicting the buffer you are reading from to swap because you need to write to new buffers, which will be assigned to the input space of the new pass. When you do the write to output at the end of the space, the whole thrash starts again ... that is not optimal.

But I think that is what is done, looking at the kernel stats as things are going n.

Oh,  just one more thing (c.f. Detective Colombo!) .. what is the correct way to synthesize incrementally, by parts/components/submodules?

I have asked to no avail. My guess is to output jsons for the subs, edit them into shape, then use them as inputs for final assembly. It is what I have tried, and it synthesizes, but I don't know if it is correct. Yes, I am aware that it will leave wires in that could be optmized away, but I believe from wat I can see that the naming conventions are such (no collisions between submodules)  that nothing gets wrongly connected up.
   
Thanks for any light that is shed ...

pbreuer, 2023-01-03:
You already have the code. It's in minisat/system.cc :

static inline int memReadPeak(void)
{
    char  name[256];
    pid_t pid = getpid();

    sprintf(name, "/proc/%d/status", pid);
    FILE* in = fopen(name, "rb");
    if (in == NULL) return 0;

    // Find the correct line, beginning with "VmPeak:":
    int peak_kb = 0;
    while (!feof(in) && fscanf(in, "VmPeak: %d kB", &peak_kb) != 1)
        while (!feof(in) && fgetc(in) != '\n')
            ;
    fclose(in);

    return peak_kb;
}

Go for it!

jix, 2023-01-03:
Ah, sorry, I missed that while quickly going over the lines in `/proc/$$/status`, it's only getrusage that doesn't provide a peak value for this.

I'm still opposed to only printing the peak vmsize instead of the peak rss as the vmsize can overestimate the required memory by quite a bit as mapped virtual memory doesn't have to be backed by RAM and how much that happens can be influenced by things like e.g. the allocator yosys was compiled with or whether anything in yosys including dependencies mmaps files etc. I wouldn't mind printing both values though.

pbreuer, 2023-01-04:
I don't think an unbacked (claimed!) mmap will be counted in peak VM, but I doubt many could say for sure! That would take quite some interrogation of the kernel code to find out. But on principle it should be counting only dirtied pages (ones you have claimed _and_ written to)

I had a look at what seemed to me to be your allocator in kernel/mem.cc and it looks to me as though you can count things (ports, wires, cells, etc) in and count them out  and that should account for everything but stack and program text (constants, statics) and the difference would be trivial compared with total memory use in a context where anyone is using enough to care. But it's way too much bother when one has that VMPeak number in /proc/self/status so convenient.

Yes, sure, put up both RSSpeak and VMpeak and  the user  will have the info.

pbreuer, 2023-01-04:
Here's a simple patch from yesterday (untested, except that it compiled and mixed fine with existing debian post-patches that hit very close by). It adds the static inline (!) memReadPeak() function to the kernel/driver.cc file containing Main() and uses it in the printout sequence, all only if __linux__ is defined.

[show_VM_peak.txt](https://github.com/YosysHQ/yosys/files/10341959/show_VM_peak.txt)