YosysHQ/yosys #3603: Assertion failed when no GHDL plugin support

suarezvictor, 2022-12-29:
### Version

Yosys 0.24+10 (git sha1 3ebc50dee, clang 11.0.1-2 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

compile yosys without GHDL support (set this in Makefile) ```ENABLE_GHDL := 0```
execute ```yosys -m ghdl```

### Expected Behavior

when the assertion is avoided (by patching the sources to return from the function instead of failing) it gives the following error:
`ERROR: This version of Yosys is built without GHDL support.`

This is prefereable, but it would be best to indicate that you should set `ENABLE_GHDL := 0` in Makefile

### Actual Behavior

Assertion fails
```ERROR: Assert `pass_register.count(pass_name) == 0' failed in kernel/register.cc:118.```

Main problem with this is that it's too difficult to know what's the problematic pass

mmicko, 2023-01-10:
Reason for assertion you get is that you still have ghdl plugin source in your tree (so it is compiled in with yosys) with pass added, but only returning error and then loading plugin as shared library tries to override pass, which is not supported. Now you will get more reasonable error for this. 
Compiling in ghdl plugin by adding it to yosys tree is bad idea, and it should be done always as shared library.

closed 2023-01-10 by mmicko

suarezvictor, 2023-01-10:
glad to see it pached, many thanks!