YosysHQ/yosys #3609: Fails to merge common logic

dev-pts, 2023-01-05:
### Version

Yosys 0.24+10 (git sha1 583ab8167, clang 14.0.0-1ubuntu1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Sorry, if I mistaken with the form. I don't know whether it is expected behavior or not.

Overall picture is that I'd like to use many instances of the same module in different submodules. And I thought I could rely on yosys that it can find common logic in its instances and merge it into one piece, thus saving FPGA resources. And, to my surprize, it really can find some common logic and merge it together, but, unfortunately, it fails on other common parts and I don't know is it even possible to force yosys to merge it too.

Please, take a look at the sample a.v ([a.v](https://github.com/YosysHQ/yosys/files/10348371/a.txt)). It contains the simplified example of the problem. It has reseted__0 and reseted__1:
```verilog
	always @(*) begin
		reseted__0 = (~(reset)) & (setup__0);
		reseted__1 = (~(reset)) & (setup__1);
	end
```

setup__0 and setup__1:
```verilog
	always @(posedge(clk)) begin
		if (reset) begin
			setup__0 <= 1;
		end else begin
			setup__0 <= 0;
		end
		if (reset) begin
			setup__1 <= 1;
		end else begin
			setup__1 <= 0;
		end
	end
```

cnt__0 and cnt__1:
```verilog
	always @(posedge(clk)) begin
		if (reset) begin
			cnt__0 <= 1;
		end else begin
			cnt__0 <= (cnt__0) ^ (1);
		end
		if (reset) begin
			cnt__1 <= 1;
		end else begin
			cnt__1 <= (cnt__1) ^ (1);
		end
	end
```

signals. As you can see, they perform literally the same logic as their counterpart.

Now, if we execute the following commands in the yosys:
```
> read_verilog a.v
> synth
> write_verilog fpga.v
```
and see into the fpga.v ([fpga.v](https://github.com/YosysHQ/yosys/files/10348369/fpga.txt)), we see that reseted__0 and reseted__1:
```verilog
  assign reseted__0 = setup__0 & ~(reset);
...
  assign reseted__1 = reseted__0;
```
setup__0 and setup__1:
```verilog
  always @(posedge clk)
    setup__0 <= reset;
...
  assign setup__1 = setup__0;
```
are merged together.

But, it fails to merge cnt__0 and cnt__1:
```verilog
  assign _1_ = ~cnt__1;
  assign _0_ = ~cnt__0;
...
  always @(posedge clk)
    if (reset) cnt__1 <= 1'h1;
    else cnt__1 <= _1_;
  always @(posedge clk)
    if (reset) cnt__0 <= 1'h1;
    else cnt__0 <= _0_;
```

Is it expected behaviour? Will it be ever possible to merge such things too?

### Expected Behavior

Everything is merged

### Actual Behavior

Something is not merged

nakengelhardt, 2023-01-06:
This is expected behavior: discovering equivalence between sequential elements with feedback loops is very complex. There are two non-default options for two passes that may sometimes be able to optimize these cases: `opt -sat` and `abc -dff`. If you want to try these out, you can create a custom script modeled after `synth`, adding these options to the `opt` call after `fsm` and/or the `abc` call after `techmap`. For larger designs this will however have a significant effect on execution time.

dev-pts, 2023-01-06:
Ok, thanks.

Using `opt -sat` and `abc -dff` did not help. Perhaps I used them wrong. Not gonna go deeper :)

closed 2023-01-06 by dev-pts