YosysHQ/yosys #3614: techmap pass result is not unique

lzxqaq, 2023-01-09:
### Feature Description

Hi, I found that techmap pass uses topological sorting, resulting in techmap pass results that are not unique. Is there any way to make techmap's results unique?

lzxqaq, 2023-01-09:
// passes/techmap/techmap.cc
```		
TopoSort<RTLIL::Cell*, IdString::compare_ptr_by_name<RTLIL::Cell>> cells;
		cells.sort();

		for (auto cell : cells.sorted)
		{
                //......
```

nakengelhardt, 2023-01-09:
What do you mean by unique?

lzxqaq, 2023-01-10:
I performed techmap pass on two identical designs one after the other, and found that their structures are different.

lzxqaq, 2023-01-10:
Here is my ys script:
```
read -sv top.v
hierarchy -check -top top
techmap
write_verilog top_techmap.v
aigmap
write_aiger top.aig

read -sv copy.v
hierarchy -check -top copy
techmap
write_verilog copy_techmap.v
aigmap
write_aiger copy.aig

// yosys-abc -c "cec top.aig copy.aig"
```

lzxqaq, 2023-01-10:
When I used yosys-abc to verify the two designs after techmap pass, I found that their structures were not equal.

lzxqaq, 2023-01-11:
I find that the root cause of this problem is that the index of `IdString` is not ordered. When you sort by `IdString`, it's not stable at all. Look at the two functions in `rtlil.h`, why use `sort_by_name_id` and not `sort_by_name_str`?

lzxqaq, 2023-01-13:
solution: #defind YOSYS_SORT_ID_FREE_LIST

closed 2023-01-13 by lzxqaq