YosysHQ/yosys #3618: Incorrect output when `$clog` used in nested function calls

robertszczepanski, 2023-01-10:
### Version

Yosys 0.25+3 (git sha1 2677569d4, gcc 9.3.0-22 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

`top.sv`:
```verilog
module top (o);
	output wire signed [31:0] o;
	parameter signed [31:0] X = 3;
	function automatic signed [$clog2(X) - 1:0] narrow;
		input reg signed [$clog2(X) - 1:0] inp;
		narrow = inp;
	endfunction
	function automatic signed [31:0] to_int;
		input reg signed [31:0] inp;
		to_int = inp;
	endfunction
	assign o = to_int(narrow(X));
endmodule
```
Executed commands:
```
$ yosys
yosys> read_verilog top.sv
```

This issue doesn't occur when `to_int` is removed or when `$clog2` is replaced by another function which just returns a value like:
```verilog
	function automatic signed [1:0] get_3;
		get_3 = 3;
	endfunction
```
so use of `$clog2` might be relevant here. 

### Expected Behavior

Produced `o` should be 32-bit wide and its value should be `3`, e.g.:
```
output signed [31:0] o;
assign o = 3;
```

### Actual Behavior

Produced `o` is 32-bit wide but its value is `-1`, generated `-dump_vlog2`:
```
Dumping Verilog AST after simplification:
    module top(o);
      /** AST_PARAMETER **/
      output signed [31:0] o;
      /** AST_FUNCTION **/
      /** AST_FUNCTION **/
      assign o = -1;
    endmodule
--- END OF AST DUMP ---
```

zachjs, 2023-01-11:
I believe Yosys's behavior on this particular example is standards-compliant. Commercial simulators and iverilog yield -1, too.

`narrow(3)` is effectively `2'(3)`, which is `2'sb11`. `to_int(2'sb11)` then simply sign-extends, yielding a regular size-32 -1.

Perhaps you have found an issue, but this example doesn't quite replicate it?

robertszczepanski, 2023-01-11:
I analyzed this example once again and it looks like I somehow misinterpreted `$clog2(3)` to be 3 instead of 2 which confused me because I was getting different results for "the same value". `-1` is of course correct, sorry for the confusion.

closed 2023-01-11 by robertszczepanski