YosysHQ/yosys #3631: Does yosys support synthesis for multi-output gates ?

ratul619, 2023-01-18:
### Feature Description

Hi , 

I have a 8 bit adder RTL code for full adder.  and my library contains a full adder cell.  
But yosys is not able to map to the full adder cell which has output pins as Sum and Cout. 

I am getting a warning "ABC: Warning: Detected 9 multi-output gates (for example, "sky130_fd_sc_hd__fa_1").
" -- Does this mean the tool is not able to support multi output gates ? 

Thanks

nakengelhardt, 2023-01-18:
Yosys does, but abc does not. Any multi-output primitive needs to be custom mapped in advance of the general logic mapping (examples of this would be BRAM or DSPs in the FPGA flows). This looks like you're using the liberty flow, where the logic mapping is entirely done by abc, so multi-output gates won't be used.

tcal-x, 2023-01-18:
@ratul619 you may want to look at the Yosys command `extract_fa`, and then `techmap` for doing a 1:1 custom mapping from `$fa` cells to the particular cells in your library.

ratul619, 2023-01-19:
Thanks @tcal-x  , works

ratul619, 2023-02-05:
Used techmap command , but the netlist which is generated fails logic equivalence.  Provided details on script used : 

https://github.com/YosysHQ/yosys/issues/3660 

Is there any way other than techmap to infer adders ? Thanks

ratul619, 2023-02-05:
@nakengelhardt  -- is there any way to use the yosys flow (not using abc) to infer the FA cells ?

povik, 2024-08-13:
> Is there any way other than techmap to infer adders ?

In one way or another there must be a techmap rule mapping to the full-adder/half-adder technology cell. Usually this rule maps the `$ha`/`$fa` internal Yosys cells, and those will appear in the netlist either through `extract_fa` or a techmap rule mapping a coarse arithmetic operation (for example `$add`, `$lcu` or `$lt`). If you have follow-up questions please open a [discussion thread](https://github.com/YosysHQ/yosys/discussions).

closed 2024-08-13 by povik