YosysHQ/yosys #3633: Regarding removal of buffers and directly using the nets

ratul619, 2023-01-19:
### Feature Description

 
Hi , 

I have buffers in my netlist , any way not to use the buffers and use the input nets of the buffer directly downstream ? 

If i put dont use on the buffers , i am getting Error that buffers are not there in the library.

sky130_fd_sc_hd__buf_1 _43_ (
    .A(_17_),
    .X(_13_)
  );
  sky130_fd_sc_hd__buf_1 _44_ (
    .A(_14_),
    .X(_10_)
  );
  sky130_fd_sc_hd__buf_1 _45_ (
    .A(_11_),
    .X(_07_)
  );

Thanks

gatecat, 2023-01-19:
What Yosys script are you using?

Ravenslofty, 2023-01-26:
The recommended solution here would be to use `techmap` followed by `opt_clean`.

Given a Verilog file like this one called `buffer_map.v`:

```verilog
module sky130_fd_sc_hd__buf_1(input A, output X)
\$_BUF_ _TECHMAP_REPLACE_ (.A(A), .Y(X));
endmodule
```

`techmap -map buffer_map.v` will replace the sky130 buffer cells with the Yosys-internal `$_BUF_` cell, and then `opt_clean` will directly connect the input and output of the `$_BUF_` cell, which achieves what you want.