YosysHQ/yosys #3637: ERROR: Can't guess frontend for input file `MACRO=32' (missing -f option)!

roby2014, 2023-01-20:
### Version

Yosys 0.24+10 (git sha1 69cbef966, gcc 12.2.0 -march=x86-64 -mtune=generic -O2 -fno-plt -fexceptions -fstack-clash-protection -fcf-protection -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

```bash
yosys -D INIT_MEM=1 DATA_W=32 ADDR_W=32 BOOTROM_ADDR_W=12 SRAM_ADDR_W=15 FIRM_ADDR_W=15 DCACHE_ADDR_W=24 N_SLAVES=1 E=31 P=30 B=29 UART=0 N_SLAVES_W=0 USE_MUL_DIV=1 USE_COMPRESSED=1
```

### Expected Behavior

Yosys REPL

### Actual Behavior

```bash
❯ yosys -D INIT_MEM=1 DATA_W=32 ADDR_W=32 BOOTROM_ADDR_W=12 SRAM_ADDR_W=15 FIRM_ADDR_W=15 DCACHE_ADDR_W=24 N_SLAVES=1 E=31 P=30 B=29 UART=0 N_SLAVES_W=0 USE_MUL_DIV=1 USE_COMPRESSED=1

 Yosys 0.24+10 (git sha1 69cbef966, gcc 12.2.0 -march=x86-64 -mtune=generic -O2 -fno-plt -fexceptions -fstack-clash-protection -fcf-protection -fPIC -Os)

-- Running command `read -define INIT_MEM=1' --
ERROR: Can't guess frontend for input file `DATA_W=32' (missing -f option)!
```

However, if I add the `-f` option, I get another error:
```bash
❯ yosys -D INIT_MEM=1 DATA_W=32 ADDR_W=32 BOOTROM_ADDR_W=12 SRAM_ADDR_W=15 FIRM_ADDR_W=15 DCACHE_ADDR_W=24 N_SLAVES=1 E=31 P=30 B=29 UART=0 N_SLAVES_W=0 USE_MUL_DIV=1 USE_COMPRESSED=1 -f verilog

 Yosys 0.24+10 (git sha1 69cbef966, gcc 12.2.0 -march=x86-64 -mtune=generic -O2 -fno-plt -fexceptions -fstack-clash-protection -fcf-protection -fPIC -Os)

-- Running command `read -define INIT_MEM=1' --

-- Parsing `DATA_W=32' using frontend `verilog' --
ERROR: Can't open input file `DATA_W=32' for reading: No such file or directory
```

If I run yosys repl and add it one by one
```
yosys> read -define INIT_MEM=1

yosys> read -define DATA_W=32
```

works fine.

Ravenslofty, 2023-01-21:
`-D` only applies for one definition at a time, so `DATA_W=32` is parsed as a file name, along with the other definitions. That's why you see `read_verilog -define INIT_MEM=1` in the log, since that got parsed as the define you intended.

To fix this, you need to specify `-D` for every definition: `yosys -D INIT_MEM=1 -D DATA_W=32 -D ADDR_W=32 -D BOOTROM_ADDR_W=12 -D SRAM_ADDR_W=15 -D FIRM_ADDR_W=15 -D DCACHE_ADDR_W=24 -D N_SLAVES=1 -D E=31 -D P=30 -D B=29 -D UART=0 -D N_SLAVES_W=0 -D USE_MUL_DIV=1 -D USE_COMPRESSED=1`

roby2014, 2023-01-22:
Thanks, that makes sense.
I solved it by doing:
```
read -define macro1=... macro2=...
```
inside a yosys script.

closed 2023-01-23 by roby2014