YosysHQ/yosys #3638: Support set_dont_touch

maliberty, 2023-01-21:
### Feature Description

Currently the only way to get yosys to not use a cell is to modify the .lib to have "dont_use : true".  It would be much cleaner if there was a set_dont_touch command with equivalent functionality.  

In OpenRoad-flow-scripts we could eliminate the need for https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/util/markDontUse.py

QuantamHD, 2023-01-23:
You can't implement this functionality in Yosys. ABC is actually the piece of software that needs this feature.

gatecat, 2023-01-23:
It would theoretically be possible to support this in Yosys by filtering the liberty files before feeding them to ABC, but I don't know whether it's worth the complexity. Yosys would also need to know about `dont_use` itself for `dfflibmap`, it's not just ABC that's touching the liberty files.

maliberty, 2023-01-24:
The complexity is now pushed onto the clients of yosys as shown in the script I linked above.  I think it better to put it in one place.  If abc needs enhancing to make this easier then we should consider that as well.

QuantamHD, 2023-01-24:
I think ABC is more than willing to take the change someone just needs to write it.

@gatecat agreed on the dfflibmap issue. I think that should be fairly easy to add given some kind of naming filter.

In advanced ASIC nodes you often have 0.5X drive strength cells you aren't supposed to use in synthesis, because their small size makes your layout more dense than can actually be supported when the cells are resized later which is why this functionality would be great to have.

QuantamHD, 2023-08-16:
Yup I added that, and wrote PR in Yosys to expose it https://github.com/YosysHQ/yosys/pull/3892