YosysHQ/yosys #3639: Techmap to yosys/examples cmos counter  using custom 74LVC** IC.

ibkvictor, 2023-01-22:
### Discussed in https://github.com/YosysHQ/yosys/discussions/3626

<div type='discussions-op-text'>

<sup>Originally posted by **ibkvictor** January 14, 2023</sup>
Hi, 

I am trying to map my design logic which uses an 14-bit counter to tech description in the liberty file. However, yosys does not recognize the tech description in the liberty file. I would appreciate any help. Thank you.

PS: I am new to yosys.
```
Extracted 0 gates and 0 wires to a netlist network with 0 inputs and 0 outputs.
Don't call ABC as there is nothing to map.
Removing temp director
...
Executing DFFLIBMAP pass (mapping DFF cells to sequential cells from liberty file).
  cell DFF (noninv, pins=3, area=18.00) is a direct match for cell type $_DFF_P_.
  cell COUNTER (noninv, pins=16, area=30.00) is a direct match for cell type $_DFF_PP0_.
  cell DFFSR (noninv, pins=5, area=18.00) is a direct match for cell type $_DFFSR_PPP_.
  final dff cell mappings:
    unmapped dff cell: $_DFF_N_
    \DFF _DFF_P_ (.C( C), .D( D), .Q( Q));
    unmapped dff cell: $_DFF_NN0_
    unmapped dff cell: $_DFF_NN1_
    unmapped dff cell: $_DFF_NP0_
    unmapped dff cell: $_DFF_NP1_
    unmapped dff cell: $_DFF_PN0_
    unmapped dff cell: $_DFF_PN1_
    \COUNTER _DFF_PP0_ (.C( C), .Q0( Q), .Q1( Q), .Q10( Q), .Q11( Q), .Q12( Q), .Q13( Q), .Q2( Q), .Q3( Q), .Q4( Q), .Q5( Q), .Q6( Q), .Q7( Q), .Q8( Q), .Q9( Q), .R( R));
    unmapped dff cell: $_DFF_PP1_
    unmapped dff cell: $_DFFSR_NNN_
    unmapped dff cell: $_DFFSR_NNP_
    unmapped dff cell: $_DFFSR_NPN_
    unmapped dff cell: $_DFFSR_NPP_
    unmapped dff cell: $_DFFSR_PNN_
    unmapped dff cell: $_DFFSR_PNP_
    unmapped dff cell: $_DFFSR_PPN_
    \DFFSR _DFFSR_PPP_ (.C( C), .D( D), .Q( Q), .R( R), .S( S));
...
     counter                         2

   Area for cell type $not is unknown!
   Area for cell type $and is unknown!
   Area for cell type $or is unknown!
   Area for cell type $mux is unknown!
   Area for cell type $dff is unknown!
   Area for cell type \counter is unknown!
```

the yosys TCL script
```
# Read verilog description of cells
read_verilog counter.v

# techmap
techmap -map counter.v counter

# mapping logic to counter.lib
abc -liberty counter.lib
dfflibmap -liberty counter.lib

# print stats
stat -liberty counter.lib

# Write out in spice format
write_spice synthesized_output.sp
```

counter.v
```
 24 module counter(clk, reset, out);
 25 
 26   parameter WIDTH = 8192;
 27 
 28   output [WIDTH-1 : 0] out;
 29   input              clk, reset;
 30 
 31   reg [WIDTH-1 : 0]   out;
 32   wire               clk, reset;
 33 
 34   always @(posedge clk or posedge reset)
 35     if (reset)
 36       out <= 0;
 37     else
 38       out <= out + 1;
```
counter.lib
```
  cell(COUNTER) {
    area: 30;
    ff("IQ", "IQN") { clocked_on: C;
                  next_state: Q0;
                       clear: R; }
    ff("IQ", "IQN") { clocked_on: Q0;
                  next_state: Q1;
                       clear: R; }
    ff("IQ", "IQN") { clocked_on: Q1;
                  next_state: Q2;
                       clear: R; }
    ff("IQ", "IQN") { clocked_on: Q2;
                  next_state: Q3;
                       clear: R; }
    ff("IQ", "IQN") { clocked_on: Q3;
                  next_state: Q4;
                       clear: R; }
    ff("IQ", "IQN") { clocked_on: Q4;
                  next_state: Q5;
                       clear: R; }
    ff("IQ", "IQN") { clocked_on: Q5;
                  next_state: Q6;
                       clear: R; }
    ff("IQ", "IQN") { clocked_on: Q6;
                  next_state: Q7;
                       clear: R; }
    ff("IQ", "IQN") { clocked_on: Q7;
                  next_state: Q8;
                       clear: R; }
    ff("IQ", "IQN") { clocked_on: Q8;
                  next_state: Q9;
                       clear: R; }
    ff("IQ", "IQN") { clocked_on: Q9;
                  next_state: Q10;
                       clear: R; }
    ff("IQ", "IQN") { clocked_on: Q10;
                  next_state: Q11;
                       clear: R; }
    ff("IQ", "IQN") { clocked_on: Q11;
                  next_state: Q12;
                       clear: R; }
    ff("IQ", "IQN") { clocked_on: Q12;
                  next_state: Q13;
                       clear: R; }
    pin(C) { direction: input;
                 clock: true; }
    pin(Q0) { direction: output;
              function: "IQ"; }
    pin(Q1) { direction: output;
              function: "IQ"; }             
    pin(Q2) { direction: output;
              function: "IQ"; }
    pin(Q3) { direction: output;
              function: "IQ"; }
    pin(Q4) { direction: output;
              function: "IQ"; }
    pin(Q5) { direction: output;
              function: "IQ"; }
    pin(Q6) { direction: output;
              function: "IQ"; }
    pin(Q7) { direction: output;
              function: "IQ"; }
    pin(Q8) { direction: output;
              function: "IQ"; }
    pin(Q9) { direction: output;
              function: "IQ"; }
    pin(Q10) { direction: output;
              function: "IQ"; }
    pin(Q11) { direction: output;
              function: "IQ"; }
    pin(Q12) { direction: output;
              function: "IQ"; }
    pin(Q13) { direction: output;
              function: "IQ"; }
    pin(R) { direction: input; }
    ; // empty statement
  }
```
</div>

gatecat, 2023-01-23:
Unfortunately, this is well beyond the level of extraction capabilities in ABC, which is intended for mapping typical combinational VLSI standard cells rather than complex sequential 74xx gates.

Limitations relevant here include:
 - sequential cells (with clocked logic) are ignored
 - multi output cells are ignored
 - ABC's ability to map cells with more than 8 inputs or so is limited

ibkvictor, 2023-01-26:
Thanks @gatecat, any chance that another software could do so? Or a manipulation of abc to work on even 4-bit or just 2-bit. It would really assist in making my project feasible.