YosysHQ/yosys #3645: Yosys generates netlist using too many LUTS

roby2014, 2023-01-24:
### Version

Yosys 0.24+10 (git sha1 69cbef966, gcc 12.2.0 -march=x86-64 -mtune=generic -O2 -fno-plt -fexceptions -fstack-clash-protection -fcf-protection -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

```
git clone https://github.com/roby2014/iob-soc
cd iob-soc
make fpga-build BOARD=COLORLIGHT_5A-75E REVISION=8.0 INIT_MEM=1 RUN_EXTMEM=1
```

### Expected Behavior

```
ERROR: Unable to place cell 'system.ext_mem0.dcache.cache_memory.n_ways_block[1].line2mem_block[4].BE_FE_block[0].cache_memory.ram[2].iob_cache_mem.ram.0.1$DPRAM_COMB0', no BELs remaining to implement cell type 'TRELLIS_COMB'
```

### Actual Behavior

Bitstream generation (it compiles with quartus/vivado toolchain)

From what I can see, yosys is generating a netlist way too big and uses alot of LUTs.
How could I reduce that?
Full log files in case it helps (synth and pnr):
https://easyupload.io/m/wk2v03

Ravenslofty, 2023-01-25:
We're probably not going to be able to easily reproduce this given the firmware in your example, which depends on the python `parse` package, and a riscv64 compiler. Could you please provide pre-built firmware images?

roby2014, 2023-01-25:
hello @Ravenslofty , thanks for answering first of all!
I've created a new branch with pre-built firmware, 
```bash
git clone https://github.com/roby2014/iob-soc
cd iob-soc
git checkout pre-built-firmware-colorlight-5a-75e-v8.0
make fpga-build BOARD=COLORLIGHT_5A-75E REVISION=8.0 INIT_MEM=1 RUN_EXTMEM=1
```

This should help, if not, tell me.

lethalbit, 2023-02-01:
Seems to error out,
```
make -C hardware/fpga/colorlight/COLORLIGHT_5A-75E build
make[1]: Entering directory '/tmp/blech/iob-soc/hardware/fpga/colorlight/COLORLIGHT_5A-75E'
../../../../hardware/hardware.mk:11: ../../../../submodules/LIB/hardware/iob_merge/hardware.mk: No such file or directory
make[1]: *** No rule to make target '../../../../submodules/LIB/hardware/iob_merge/hardware.mk'.  Stop.
make[1]: Leaving directory '/tmp/blech/iob-soc/hardware/fpga/colorlight/COLORLIGHT_5A-75E'
make: *** [Makefile:95: fpga-build] Error 2
```

If at all possible could you just zip up the final output verilog and json and attach it to the issue?

That way it's a lot easier to just stuff it into Yosys and nextpnr to inspect.

lethalbit, 2023-02-03:
I suspect this is related to #3643

Ravenslofty, 2023-02-03:
Sorry for the delay; I got around to building this finally.

There are definitely things which stick out to me; even without modifying the Yosys parameters, `nextpnr-ecp5` outputs
```
Warning: IO 'trap' is unconstrained in LPF and will be automatically placed
Warning: IO 'resetn' is unconstrained in LPF and will be automatically placed
Warning: timing analysis failed due to presence of combinatorial loops, incomplete specification of timing ports, etc.
```
which should really be fixed.

Anyway, I disagree that this is related to #3643, because `-no-rw-check` makes the synthesis result *worse* rather than better.

Without `-no-rw-check`, Yosys outputs
```
[00080.628510]    Number of cells:              39656
[00080.628510]      CCU2C                         580
[00080.628511]      DP16KD                         24
[00080.628511]      L6MUX21                       722
[00080.628512]      LUT4                        25653
[00080.628513]      MULT18X18D                      4
[00080.628513]      PFUMX                        2855
[00080.628514]      TRELLIS_DPR16X4              4200
[00080.628515]      TRELLIS_FF                   5618
```

With `-no-rw-check`, Yosys outputs
```
[00082.577591]    Number of cells:              53067
[00082.577592]      CCU2C                         580
[00082.577592]      DP16KD                         24
[00082.577593]      L6MUX21                      2324
[00082.577593]      LUT4                        32399
[00082.577594]      MULT18X18D                      4
[00082.577594]      PFUMX                        7918
[00082.577596]      TRELLIS_DPR16X4              4200
[00082.577596]      TRELLIS_FF                   5618
```

lethalbit, 2023-02-03:
So it does! oops, sorry about that ^^;

Ravenslofty, 2023-02-03:
...Apparently I need some more tea, because actually this would happen if the RW enforcement logic being removed massively shortens the critical path, necessitating a lot of wide LUTs to keep to minimum delay. This seems fairly likely.

Ravenslofty, 2023-02-03:
Okay, so, `-abc9 -nowidelut -no-rw-check` gives me this:

```
[00101.414421]    Number of cells:              33778
[00101.414422]      CCU2C                         554
[00101.414422]      DP16KD                         24
[00101.414423]      LUT4                        23378
[00101.414424]      MULT18X18D                      4
[00101.414424]      TRELLIS_DPR16X4              4200
[00101.414425]      TRELLIS_FF                   5618
```

Unfortunately, I think this is probably *never* going to fit on the chip without design changes, because while I can spend lots and lots of time going through the various tricks I know of to reduce LUT count, this design needs 4,200 `TRELLIS_DPR16X4`s, but the ECP5-25k only has 3,036.