YosysHQ/yosys #3648: Segfault on write_firrtl

ggeier, 2023-01-25:
### Version

Yosys 0.23+45 (git sha1 a64ed824e, clang 10.0.0-4ubuntu1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

```
read_verilog ibex_top.v
hierarchy -auto-top
proc
async2sync
write_firrtl IbexTop.fir
```
[ibex_top.zip](https://github.com/YosysHQ/yosys/files/10499857/ibex_top.zip)

### Expected Behavior

Write the design to `IbexTop.fir`

### Actual Behavior

```
yosys> write_firrtl IbexTop.fir

5. Executing FIRRTL backend.

5.1. Executing PMUXTREE pass.

5.2. Executing BMUXMAP pass.

5.3. Executing DEMUXMAP pass.
Segmentation fault (core dumped)
```
The target file remains empty.

lethalbit, 2023-01-25:
It looks like in [`firrtl.cc:emit_elaborated_extmodules`](https://github.com/YosysHQ/yosys/blob/master/backends/firrtl/firrtl.cc#L336) the result from the call to [`design->module(cell->type);`](https://github.com/YosysHQ/yosys/blob/master/backends/firrtl/firrtl.cc#L348) returns NULL and always assumes that it's valid

I'm not too familiar with the internals myself but I don't think that's correct, I'll bring this up at the next dev meeting if no one else from the team chimes in before then.

nakengelhardt, 2023-01-30:
The primary cause here is that there are unknown modules instantiated in the design (if you add `-check` to the call to `hierarchy` it errors out right away), but `write_firrtl` should handle that more gracefully.

commit 430b9cb9 (2023-02-01): backends/firrtl: Ensure `modInstance` is valid

This should fix #3648 where when calling `emit_elaborated_extmodules` it
checks to see if a module is a black-box, however there was no
validation that the cell type was actually known, and it just always
assumed that we would get a valid instance, causing a segfault.
commit 8489ef34 (2023-02-01): backends/firrtl: Ensure `modInstance` is valid

This should fix #3648 where when calling `emit_elaborated_extmodules` it
checks to see if a module is a black-box, however there was no
validation that the cell type was actually known, and it just always
assumed that we would get a valid instance, causing a segfault.
PR #3656 (2023-02-03): backends/firrtl: Ensure `modInstance` is valid
This should fix #3648 where when calling `emit_elaborated_extmodules` it checks to see if a module is a black-box, however there was no validation that the cell type was actually known, and it just always assumed that we would get a valid instance, causing a segfault.
commit a90f9406 (2023-02-03): backends/firrtl: Ensure `modInstance` is valid

This should fix #3648 where when calling `emit_elaborated_extmodules` it
checks to see if a module is a black-box, however there was no
validation that the cell type was actually known, and it just always
assumed that we would get a valid instance, causing a segfault.
closed 2023-02-03 by lethalbit