YosysHQ/yosys #3657: yosys fails in synthesizing flip-flop with asynchronous reset and ternary operator ?:

astrollo, 2023-02-01:
### Version

Yosys 0.25+86 (git sha1 541fdffff, clang 10.0.0-4ubuntu1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Let us consider this simple code:
```
module ff2 (input logic d, r, ck,
   output logic q);
always_ff @ (posedge ck , posedge r)
 q <= r ? '0 : d;
endmodule  
```

### Expected Behavior

A flip-flop with asynchronous reset should be inferred. 

### Actual Behavior

yosys exit after proc command:
```
2. Executing PROC pass (convert processes to netlists).

2.1. Executing PROC_CLEAN pass (remove empty switches from decision trees).
Cleaned up 0 empty switches.

2.2. Executing PROC_RMDEAD pass (remove dead branches from decision trees).
Removed a total of 0 dead cases.

2.3. Executing PROC_PRUNE pass (remove redundant assignments in processes).
Removed 0 redundant assignments.
Promoted 1 assignment to connection.

2.4. Executing PROC_INIT pass (extract init attributes).

2.5. Executing PROC_ARST pass (detect async resets in processes).

2.6. Executing PROC_ROM pass (convert switches to ROMs).
Converted 0 switches.

2.7. Executing PROC_MUX pass (convert decision trees to multiplexers).
Creating decoders for process `\ff2.$proc$ff2.sv:4$1'.

2.8. Executing PROC_DLATCH pass (convert process syncs to latches).

2.9. Executing PROC_DFF pass (convert process syncs to FFs).
Creating register for signal `\ff2.\q' using process `\ff2.$proc$ff2.sv:4$1'.
ERROR: Multiple edge sensitive events found for this signal!
```

Ravenslofty, 2023-02-03:
I will point out that if you write the flop like this:

```systemverilog
always_ff @(posedge ck or posedge r)
  if (r)
    q <= '0;
  else
    q <= d;
```

similar to IEEE 1364.1-2002 example 12 (under section 5.2.2.1), then Yosys will correctly infer the flop:

```
2.9. Executing PROC_DFF pass (convert process syncs to FFs).
Creating register for signal `\ff2.\q' using process `\ff2.$proc$test.sv:3$1'.
  created $adff cell `$procdff$2' with positive edge clock and positive level reset.
```

Alternatively, Verific in Tabby CAD Suite can infer the flop from your source code just fine.

whitequark, 2023-02-03:
Is this actually a bug? The exact set of patterns that are recognized in processes for flops with reset isn't specified anywhere in the Verilog spec, so it's up to Yosys to decide and document which ones are. If I recall it's not currently documented in the manual.

astrollo, 2023-02-10:
It's not exactly a bug, but rather a limitation related specifically to the inference of asynchronous reset. Different uses of the ternary operator in always_ff blocks are correctly synthesized, as shown in the following example:
```
always_ff @ (posedge ck, posedge r)
 if (r)
   q <= '0;
 else
  q <= en? d : q;
```