YosysHQ/yosys #3660: Equivalence fails when using full adder mapping

ratul619, 2023-02-04:
### Version

Yosys 0.23+45 (git sha1 a64ed824e, clang 10.0.0-4ubuntu1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Hi , 

Getting equivalnce issues when checking in between RTL and netlist. The netlist is generated using mapping and inferencing full adders : 

Input RTL : rippler_adder.v

module ripplemod(a, b, cin, sum, cout);

input [7:0] a;

input [7:0] b;

input cin;

output [7:0]sum;

output cout;

wire[6:0] c;

fulladd a1(a[0],b[0],cin,sum[0],c[0]);

fulladd a2(a[1],b[1],c[0],sum[1],c[1]);

fulladd a3(a[2],b[2],c[1],sum[2],c[2]);

fulladd a4(a[3],b[3],c[2],sum[3],c[3]);

fulladd a5(a[4],b[4],c[3],sum[4],c[4]);

fulladd a6(a[5],b[5],c[4],sum[5],c[5]);

fulladd a7(a[6],b[6],c[5],sum[6],c[6]);

fulladd a8(a[7],b[7],c[6],sum[7],cout);

endmodule

 

module fulladd(a, b, cin, sum, cout);

input a;

input b;

input cin;

output sum;

output cout;

assign sum= (a&!b&!cin) | (!a&b&!cin) | (!a&!b&cin) | (a&b&cin);

//(A&!B&!CI) | (!A&B&!CI) | (!A&!B&CI) | (A&B&CI)

assign cout=((a&b)|(b&cin)|(a&cin));

endmodule

----------------------------------------------
Yosys Script : 

# read design
read -sv  rippler_adder.v
hierarchy -top ripplemod
proc; fsm; opt; memory; opt

synth -flatten; 

        extract_fa -fa -v
        extract_fa -ha -v
       
techmap -map /pdks/sky130A/libs.tech/openlane/sky130_fd_sc_hd/fa_map.v

dfflibmap -liberty ./sky130_fd_sc_hd__tt_025C_1v80.lib 

# mapping logic to mycells.lib
abc -liberty  ./sky130_fd_sc_hd__tt_025C_1v80.lib

opt_clean -purge

write_verilog RA_synth_FV.v

------------------------------------------------------------------------------------------------

11. Executing EQUIV_SIMPLE pass.
Found 9 unproven $equiv cells (9 groups) in equiv:
  Trying to prove $equiv for \cout: success!
  Trying to prove $equiv for \sum[0]: failed.
  Trying to prove $equiv for \sum[3]: failed.
  Trying to prove $equiv for \sum[2]: failed.
  Trying to prove $equiv for \sum[1]: failed.
  Trying to prove $equiv for \sum[4]: failed.
  Trying to prove $equiv for \sum[7]: success!
  Trying to prove $equiv for \sum[6]: failed.
  Trying to prove $equiv for \sum[5]: failed.
Proved 2 previously unproven $equiv cells.

12. Executing EQUIV_INDUCT pass.
Found 7 unproven $equiv cells in module equiv:
  Proving existence of base case for step 1. (2832 clauses over 1104 variables)
  Proof for base case failed. Circuit inherently diverges!
Proved 0 previously unproven $equiv cells.

13. Executing EQUIV_STATUS pass.
Found 9 $equiv cells in equiv:
  Of those cells 2 are proven and 7 are unproven.
  Unproven $equiv $auto$equiv_make.cc:242:find_same_wires$39: \a7.sum_gold \_37_.Y_gate
  Unproven $equiv $auto$equiv_make.cc:242:find_same_wires$40: \a6.sum_gold \_36_.Y_gate
  Unproven $equiv $auto$equiv_make.cc:242:find_same_wires$41: \a5.sum_gold \_35_.Y_gate
  Unproven $equiv $auto$equiv_make.cc:242:find_same_wires$42: \a4.sum_gold \_34_.Y_gate
  Unproven $equiv $auto$equiv_make.cc:242:find_same_wires$43: \a3.sum_gold \_33_.Y_gate
  Unproven $equiv $auto$equiv_make.cc:242:find_same_wires$44: \a2.sum_gold \_31_.Y_gate
  Unproven $equiv $auto$equiv_make.cc:242:find_same_wires$45: \a1.sum_gold \_57_.SUM_gate
Found a total of 7 unproven $equiv cells.

### Expected Behavior

since its just a ripple caarry adder , its expected to have only full adders connected serially.

### Actual Behavior

failing in equivalnce

ratul619, 2023-02-04:
Script for FV : 

read_verilog -defer rippler_adder.v 
prep -flatten -top ripplemod
splitnets -ports
design -stash ripplemod
read_verilog  RA_synth_FV.v 
prep -flatten -top ripplemod_synth
splitnets -ports
design -stash ripplemod_synth
design -copy-from  ripplemod -as gold ripplemod
design -copy-from ripplemod_synth -as gate ripplemod_synth
equiv_make gold gate equiv
prep -flatten -top equiv
opt_clean -purge
opt -full
equiv_simple
equiv_induct
equiv_status -assert

ratul619, 2023-02-04:
Netlist : 

/* Generated by Yosys 0.23+45 (git sha1 a64ed824e, clang 10.0.0-4ubuntu1 -fPIC -Os) */

(* hdlname = "\\ripplemod" *)
(* top =  1  *)

module sky130_fd_sc_hd__fa_1(A,B,CIN,COUT,SUM);
input A,B,CIN;
output COUT,SUM;

assign SUM = A ^B ^ CIN;
assign COUT = (A & B)  |  ( B & CIN)  | ( CIN & A);
endmodule

module  sky130_fd_sc_hd__maj3_1(A,B,C,X);
input A,B,C;
output X;

assign X = (A & B)  |  ( B & C)  | ( C & A);
endmodule

 module sky130_fd_sc_hd__inv_1 (A, Y); // An inverter
 input A;
 output Y;
 assign Y = ~A;
 endmodule

 module  sky130_fd_sc_hd__and2_0 (A, B,X); // An inverter
 input A;
 input B;
 output X;
 assign X = A & B;
 endmodule

 module  sky130_fd_sc_hd__or2_0 (A, B,X); // An inverter
 input A;
 input B;
 output X;
 assign X = A || B;
 endmodule

 module sky130_fd_sc_hd__buf_1 (A, X); // An inverter
 input A;
 output X;
 assign X = A;
 endmodule

module ripplemod_synth(a, b, cin, sum, cout);
  wire _00_;
  wire _01_;
  wire _02_;
  wire _03_;
  wire _04_;
  wire _05_;
  (* force_downto = 32'd1 *)
  wire _06_;
  (* force_downto = 32'd1 *)
  wire _07_;
  (* force_downto = 32'd1 *)
  wire _08_;
  (* force_downto = 32'd1 *)
  wire _09_;
  (* force_downto = 32'd1 *)
  wire _10_;
  (* force_downto = 32'd1 *)
  wire _11_;
  (* force_downto = 32'd1 *)
  wire _12_;
  (* force_downto = 32'd1 *)
  wire _13_;
  (* force_downto = 32'd1 *)
  wire _14_;
  (* force_downto = 32'd1 *)
  wire _15_;
  (* force_downto = 32'd1 *)
  wire _16_;
  (* force_downto = 32'd1 *)
  wire _17_;
  (* force_downto = 32'd1 *)
  wire _18_;
  (* force_downto = 32'd1 *)
  wire _19_;
  (* force_downto = 32'd1 *)
  wire _20_;
  (* force_downto = 32'd1 *)
  wire _21_;
  (* force_downto = 32'd1 *)
  wire _22_;
  (* force_downto = 32'd1 *)
  wire _23_;
  (* force_downto = 32'd1 *)
  wire _24_;
  (* force_downto = 32'd1 *)
  wire _25_;
  (* force_downto = 32'd1 *)
  wire _26_;
  (* force_downto = 32'd1 *)
  wire _27_;
  input [7:0] a;
  wire [7:0] a;
  input [7:0] b;
  wire [7:0] b;
  input cin;
  wire cin;
  output cout;
  wire cout;
  output [7:0] sum;
  wire [7:0] sum;
  sky130_fd_sc_hd__inv_1 _28_ (
    .A(cin),
    .Y(_26_)
  );
  sky130_fd_sc_hd__inv_1 _29_ (
    .A(b[0]),
    .Y(_25_)
  );
  sky130_fd_sc_hd__inv_1 _30_ (
    .A(a[0]),
    .Y(_24_)
  );
  sky130_fd_sc_hd__inv_1 _31_ (
    .A(_23_),
    .Y(sum[1])
  );
  sky130_fd_sc_hd__inv_1 _32_ (
    .A(_27_),
    .Y(_21_)
  );
  sky130_fd_sc_hd__inv_1 _33_ (
    .A(_20_),
    .Y(sum[2])
  );
  sky130_fd_sc_hd__inv_1 _34_ (
    .A(_17_),
    .Y(sum[3])
  );
  sky130_fd_sc_hd__inv_1 _35_ (
    .A(_14_),
    .Y(sum[4])
  );
  sky130_fd_sc_hd__inv_1 _36_ (
    .A(_11_),
    .Y(sum[5])
  );
  sky130_fd_sc_hd__inv_1 _37_ (
    .A(_08_),
    .Y(sum[6])
  );
  sky130_fd_sc_hd__or2_0 _38_ (
    .A(_07_),
    .B(b[7]),
    .X(_00_)
  );
  sky130_fd_sc_hd__and2_0 _39_ (
    .A(_07_),
    .B(b[7]),
    .X(_01_)
  );
  sky130_fd_sc_hd__or2_0 _40_ (
    .A(a[7]),
    .B(_00_),
    .X(_02_)
  );
  sky130_fd_sc_hd__maj3_1 _41_ (
    .A(_07_),
    .B(a[7]),
    .C(b[7]),
    .X(cout)
  );
  sky130_fd_sc_hd__inv_1 _42_ (
    .A(cout),
    .Y(_03_)
  );
  sky130_fd_sc_hd__and2_0 _43_ (
    .A(_02_),
    .B(_03_),
    .X(_04_)
  );
  sky130_fd_sc_hd__and2_0 _44_ (
    .A(a[7]),
    .B(_01_),
    .X(_05_)
  );
  sky130_fd_sc_hd__or2_0 _45_ (
    .A(_04_),
    .B(_05_),
    .X(sum[7])
  );
  sky130_fd_sc_hd__buf_1 _46_ (
    .A(_22_),
    .X(_18_)
  );
  sky130_fd_sc_hd__buf_1 _47_ (
    .A(_19_),
    .X(_15_)
  );
  sky130_fd_sc_hd__buf_1 _48_ (
    .A(_16_),
    .X(_12_)
  );
  sky130_fd_sc_hd__buf_1 _49_ (
    .A(_13_),
    .X(_09_)
  );
  sky130_fd_sc_hd__buf_1 _50_ (
    .A(_10_),
    .X(_06_)
  );
  sky130_fd_sc_hd__fa_1 _51_ (
    .A(a[6]),
    .B(b[6]),
    .CIN(_06_),
    .COUT(_07_),
    .SUM(_08_)
  );
  sky130_fd_sc_hd__fa_1 _52_ (
    .A(a[5]),
    .B(b[5]),
    .CIN(_09_),
    .COUT(_10_),
    .SUM(_11_)
  );
  sky130_fd_sc_hd__fa_1 _53_ (
    .A(a[4]),
    .B(b[4]),
    .CIN(_12_),
    .COUT(_13_),
    .SUM(_14_)
  );
  sky130_fd_sc_hd__fa_1 _54_ (
    .A(a[3]),
    .B(b[3]),
    .CIN(_15_),
    .COUT(_16_),
    .SUM(_17_)
  );
  sky130_fd_sc_hd__fa_1 _55_ (
    .A(a[2]),
    .B(b[2]),
    .CIN(_18_),
    .COUT(_19_),
    .SUM(_20_)
  );
  sky130_fd_sc_hd__fa_1 _56_ (
    .A(a[1]),
    .B(b[1]),
    .CIN(_21_),
    .COUT(_22_),
    .SUM(_23_)
  );
  sky130_fd_sc_hd__fa_1 _57_ (
    .A(_24_),
    .B(_25_),
    .CIN(_26_),
    .COUT(_27_),
    .SUM(sum[0])
  );
endmodule

gatecat, 2023-02-09:
How are you running the equivalence checking? Have you told it about the contents of `sky130_fd_sc_hd__fa_1` anyhow (for example with a `read_liberty` without `-lib`?)

Sysynth, 2023-04-02:
If I'm not mistaken, extract_fa uses the same cell (`$fa`) for both full adders and half adders. 

In the script you shared, you are extracting the full adders and half adders from the design and then mapping all of them to the same full adder cell from the `fa_map.v` file, leading to unexpected behavior. Thus, if you are trying to have 8 full adders connected serially, then you should remove the `extract_fa -ha -v` command from your script.