YosysHQ/yosys #3666: Several tests fail

yurivict, 2023-02-10:
### Version

0.26

### On which OS did this happen?

BSD

### Reproduction Steps

'make test'

### Expected Behavior

n/a

### Actual Behavior

See [log](https://people.freebsd.org/~yuri/yosys-0.26-test.log).

gatecat, 2023-02-10:
I think the only actual failure is the `make[2]: don't know how to make w. Stop` at the end (which definitely looks like a real problem in the tests somehow).

The other errors all appear to be part of tests that are checking error generation in some way and so are expected to produce those errors.

jix, 2023-02-10:
Some new tests were inadvertently invoking `make` directly instead of using the `MAKE` environment variable when set, failing on systems where `make` isn't compatible with GNU make. This should be fixed in #3667.

@yurivict can you test that PR and let me know whether there are any remaining test failures? I don't have a BSD system at hand to quickly test it myself

yurivict, 2023-02-10:
The last error is gone with https://github.com/YosysHQ/yosys/pull/3667.

yurivict, 2023-02-10:
But it has many lines like this:
```
xprop_ff_3: failed
xprop_dff_1pd: failed
xprop_dff_1nd: failed
xprop_dff_3pd: failed
xprop_dff_3nd: failed
xprop_dffe_1pnd: failed
xprop_dffe_1nnd: failed
xprop_dffe_3pnd: failed
xprop_dffe_3nnd: failed
```

jix, 2023-02-10:
As @gatecat mentioned above, those syntax errors are produced as part of testing the error handling of yosys's own parser. This specific test for example `tests/various/bug1745.ys` contains the line `logger -expect error "syntax error, unexpected TOK_CONSTVAL" 1` so this is indeed explicitly tested for. The test suite would fail when that error is _not_ generated, as that would be a bug. You can see the same error messages when the tests are running on the CI: https://github.com/YosysHQ/yosys/actions/runs/4144601154/jobs/7167858083 

For `xprop` the situation is slightly different, `xprop` is experimental and test failures for this pass specifically are ignored as it's not yet able to reliably pass everything the tests are testing for. There seems to be some issue that causes all tests to fail for you, while I'm expecting just a few failures, but as this behavior is reproducible on our Linux CI I should be able to sort that out. It does not indicate any issue with your build.

jix, 2023-02-13:
The remaining issues with the failing xprop tests this highlighted were fixed in #3669.

closed 2023-02-13 by jix