YosysHQ/yosys #3679: ice40: BRAM is replaced with DFFs even when conditionally assigned read buffer is exhaustive

dirkenstein, 2023-02-17:
### Version

Yosys 0.26+34 (git sha1 a20804c6e, clang 13.0.0 -fPIC -Os)

### On which OS did this happen?

macOS

### Reproduction Steps

The following verilog  sythesizes using BRAM on ice40:

```
   reg [31:0] ram[0:(mem_size*256)-1];
   reg [31:0] ram_read;

   wire [$clog2(mem_size*256):0] ram_read_addr;

   always @(posedge clk) begin
     if (prog_mode) begin
        ram_read_addr <= prog_addr >> 1;
     end else if (!rst_out) begin
        if (dBus_cmd_valid && !dBus_cmd_payload_wr && dbus_ram_access)
          ram_read_addr <= dBus_cmd_payload_address[15:2];
        else if (ibus_stalled)
          ram_read_addr <= ibus_stall_pc;
        else if (iBus_cmd_valid)
          ram_read_addr <= iBus_cmd_payload_pc[15:2];
     end
     ram_read <= ram[ram_read_addr];
   end
```
This uses DFFs, and exhausts the LUTs on an ice40 hx8k mem_size is 12):

```
   reg [31:0] ram[0:(mem_size*256)-1];
   reg [31:0] ram_read;

   always @(posedge clk)
     if (prog_mode) begin
        ram_read <= ram[prog_addr >> 1];
     end else if (!rst_out) begin
        if (dBus_cmd_valid && !dBus_cmd_payload_wr && dbus_ram_access)
          ram_read <= ram[dBus_cmd_payload_address[15:2]];
        else if (ibus_stalled)
          ram_read <= ram[ibus_stall_pc];
        else if (iBus_cmd_valid)
          ram_read <= ram[iBus_cmd_payload_pc[15:2]];
     end

```
This seems more or less logical  given that Lattice BRAM needs an output buffer to use, and it's conditionally assigned here. There are fall-through cases where the buffer isn't assigned to ram_read at all. In the overall design those conditions shouldn't actualy occur, but yosys can't  infer that, as far as i can tell. 

But how about this case, where there is no way to leave the nested if statements without having assigned a value to ram_read?

 ```
 reg [31:0] ram[0:(mem_size*256)-1];
  reg [31:0] ram_read;

 always @(posedge clk)
     if (prog_mode) begin
        ram_read <= ram[prog_addr >> 1];
     end else if (!rst_out) begin
        if (dBus_cmd_valid && !dBus_cmd_payload_wr && dbus_ram_access)
          ram_read <= ram[dBus_cmd_payload_address[15:2]];
        else if (ibus_stalled)
          ram_read <= ram[ibus_stall_pc];
        else if (iBus_cmd_valid)
          ram_read <= ram[iBus_cmd_payload_pc[15:2]];
        else
          ram_read <= ram[0];
     end else
       ram_read <= ram[0];
```
Yosys doesn't seem to be able to infer that there is no pssible outcome of the if statements that leaves ram_read (the output buffer) unassigned. 

Is this intended/expected behaviour? 0.17 actually used BRAM for the first example, too.

### Expected Behavior

Yosys synthesizes the ram using BRAM blocks.

### Actual Behavior

Yosys synthesizes the ram using DFFs.

zeldin, 2023-02-17:
So, I made a minimal stand-alone reproduction recipe for this.  The non-exhaustive conditional assignment may or may not be a red herring -- as I show below it is possible to have non-exhaustive conditional assignment and still get BRAM with current Yosys.

With yosys 0.17, the following module synthesized for ice40 correctly infers BRAMs:
```
module myrom1(input         clk,
              input [13:0]  addr_in,
              input         normal_addr_read_en_in,
              input         inverted_addr_read_en_in,
              output [7:0]  data_out);

   reg [7:0] rom[0:16383];
   reg [7:0] rom_read;

   assign data_out = rom_read;

   initial
     $readmemh("rom.mem", rom);

   always @(posedge clk)
     if (normal_addr_read_en_in)
       rom_read <= rom[addr_in];
     else if (inverted_addr_read_en_in)
       rom_read <= rom[~addr_in];

endmodule // myrom1
```
The output looks correct, with muxes on the `RADDR` ports and enable logic on `RCE`.  However, with yosys 0.18 and all the way up to 0.26 I get LUT memory instead, and the design will not fit in HX8K.
In contrast, the following, sematically completely equivalent, module infers BRAMs in all yosys versions from 0.17 to 0.26:
```
module myrom2(input         clk,
              input [13:0]  addr_in,
              input         normal_addr_read_en_in,
              input         inverted_addr_read_en_in,
              output [7:0]  data_out);

   reg [7:0] rom[0:16383];
   reg [7:0] rom_read;

   assign data_out = rom_read;

   initial
     $readmemh("rom.mem", rom);

   always @(posedge clk)
     if (normal_addr_read_en_in || inverted_addr_read_en_in)
       rom_read <= rom[inverted_addr_read_en_in? ~addr_in : addr_in];

endmodule // myrom2
```

nakengelhardt, 2023-02-22:
@dirkenstein Memory inference is pattern based. If the logic produced by the code pattern matches what memory_libmap is looking for, it can map it to BRAM, otherwise not. In this case, when you put the `if` outside the array access, the mux will end up on the "wrong" side of the memory, and it can't be mapped. The [coding patterns supported by memory_libmap are listed here](https://docs.google.com/document/d/1MR1IV34XGJBIGMYu2YzjE-2KaNaP3R9jaPLj1RTU2j0/edit#heading=h.hxvjs26wvyc6).

@zeldin Those two descriptions are not semantically equivalent at all! Not only is the priority between `normal_addr_read_en_in` and `inverted_addr_read_en_in` inversed in the case where both are high, but more importantly, the behavior is different when both are low: `myrom2` will read from `addr_in` in that case, whereas `myrom1` will keep the previously read value.

nakengelhardt, 2023-02-23:
You're right, I misinterpreted the second part! Those should be equivalent (with the priority changed) as long as it's the only port.

Since 0.18 it's an entirely different pass doing memory inference (`memory_libmap` instead of `memory_bram`), so it does things quite differently in many places, but I'll see if I can figure out more about what's happening with `myrom1`.

zeldin, 2023-02-23:
@nakengelhardt Thanks.  I get that the implementation is different now, but hopefully it should be possible to make it produce the same (or better 😄 ) results.

nakengelhardt, 2023-02-24:
Well no, the whole point was that `memory_bram` would often produce results that don't match the semantics of the input code; `memory_libmap` is much more strict about correctness, and it very deliberately generates extra emulation logic, or does not map at all, when it cannot guarantee that the BRAM mapping would have identical behavior. It is entirely intended that for many examples you will have "worse" (in terms of area) results.

zeldin, 2023-02-24:
@nakengelhardt Yes, I've read your messages about this in other issues.  In this particular case though I'm fairly confident that no extra emulation logic is needed, only the address mux and RCE signal that 0.17 generates.