YosysHQ/yosys #3691: yosys fails in synthesizing D flip-flop

1353369570, 2023-02-28:
### Version

yosys -0.15

### On which OS did this happen?

Linux

### Reproduction Steps

consider the following code:
module dff(
	input clk,
	input rst_n,
	input d,
	output reg q
);
always @ (posedge clk or negedge rst_n)
begin	
	q <=!rst_n ? '0 : d;		
end
endmodule

and the following command

yosys -p 'read_verilog rtl.v; synth; write_verilog syn.v'

### Expected Behavior

A flip-flop  reset should be inferred.

### Actual Behavior

 Executing SYNTH pass.

2.1. Executing HIERARCHY pass (managing design hierarchy).

2.2. Executing PROC pass (convert processes to netlists).

2.2.1. Executing PROC_CLEAN pass (remove empty switches from decision trees).
Cleaned up 0 empty switches.

2.2.2. Executing PROC_RMDEAD pass (remove dead branches from decision trees).
Removed a total of 0 dead cases.

2.2.3. Executing PROC_PRUNE pass (remove redundant assignments in processes).
Removed 0 redundant assignments.
Promoted 1 assignment to connection.

2.2.4. Executing PROC_INIT pass (extract init attributes).

2.2.5. Executing PROC_ARST pass (detect async resets in processes).

2.2.6. Executing PROC_MUX pass (convert decision trees to multiplexers).
Creating decoders for process `\dff.$proc$2.v:8$1'.

2.2.7. Executing PROC_DLATCH pass (convert process syncs to latches).

2.2.8. Executing PROC_DFF pass (convert process syncs to FFs).
Creating register for signal `\dff.\q' using process `\dff.$proc$2.v:8$1'.
ERROR: Multiple edge sensitive events found for this signal!

nakengelhardt, 2023-02-28:
This is the same problem as #3657 (and this example has it without using SystemVerilog keywords).

abdulhameed-rs, 2023-11-01:
I am getting same error for following code:
module test(
input clk, rst, start,
input a,b,
output reg out);
wire temp;
assign temp = a & b;

always @ (posedge clk or posedge rst or posedge start)
begin
	if(rst | start)
	begin
		out <= 0;
	end
	else
	begin
		out <= temp;
	end
end
	
endmodule

**ERROR: Multiple edge sensitive events found for this signal!**

Is there any work going on to support this?

**_NOTE; Vivado is able to synthesize this code_**

nakengelhardt, 2023-11-02:
@abdulhameed-rs that is unrelated to this issue; there are actually multiple edge sensitive events in your example (`posedge rst` and `posedge start`) as is pointed out by `proc`. There is no dff type in yosys internal library that can take two independent reset signals, and using a logic or gate can lead to glitches, so if you want to do this you have to assign it explicitly to an intermediate wire to give you a chance to consider the problem and make sure this is what you want.