YosysHQ/yosys #3697: Undef propagation in power operator

1353369570, 2023-03-07:
### Version

yosys -0.15

### On which OS did this happen?

Linux

### Reproduction Steps

Consider the following test case:
module test(a, y);
  input [3:0] a;
  output [3:0] y;
  assign y = 4'd2 ** (4'd1/a);
endmodule

and the following command

yosys -p 'read_verilog rtl.v; synth; write_verilog syn.v'

 

### Expected Behavior

 y should be 4'bx when a is zero

### Actual Behavior

/* Generated by Yosys 0.15 (git sha1 UNKNOWN, gcc 9.4.0-1ubuntu1~20.04 -fPIC -Os) */

(* src = "2.v:1.1-5.10" *)
module test(a, y);
  wire _00_;
  wire _01_;
  wire _02_;
  wire _03_;
  wire _04_;
  wire _05_;
  wire _06_;
  wire _07_;
  wire _08_;
  wire _09_;
  wire _10_;
  wire _11_;
  wire _12_;
  wire _13_;
  wire _14_;
  wire _15_;
  wire _16_;
  wire _17_;
  wire _18_;
  wire _19_;
  wire _20_;
  wire _21_;
  wire _22_;
  wire _23_;
  wire _24_;
  wire _25_;
  wire _26_;
  wire _27_;
  wire _28_;
  wire _29_;
  wire _30_;
  wire _31_;
  wire _32_;
  wire _33_;
  wire _34_;
  wire _35_;
  wire _36_;
  wire _37_;
  wire _38_;
  wire _39_;
  wire _40_;
  wire _41_;
  wire _42_;
  wire _43_;
  wire _44_;
  wire _45_;
  (* src = "2.v:2.15-2.16" *)
  input [3:0] a;
  wire [3:0] a;
  (* src = "2.v:3.16-3.17" *)
  output [3:0] y;      
  wire [3:0] y;
  assign _45_ = ~(a[3] | a[2]); 
  assign _00_ = ~(a[1] | a[2]);
  assign _01_ = a[1] | a[0];
  assign _02_ = _01_ | ~(_45_);
  assign _03_ = ~a[3];
  assign _04_ = ~a[0];
  assign _05_ = ~(_00_ & _04_);
  assign _06_ = _03_ & ~(_05_);
  assign _07_ = ~(_06_ & a[0]);
  assign _08_ = ~(a[1] ^ a[0]);
  assign _09_ = _02_ ? _07_ : _08_;
  assign _10_ = _09_ | a[2];
  assign _11_ = _09_ ^ a[2];
  assign _12_ = _02_ | _04_;
  assign _13_ = _12_ | a[1];
  assign _14_ = _11_ & ~(_13_);
  assign _15_ = _10_ & ~(_14_);
  assign _16_ = ~(_12_ ^ a[1]);
  assign _17_ = _16_ | ~(_11_);
  assign _18_ = _04_ & ~(_17_);
  assign _19_ = _15_ & ~(_18_);
  assign _20_ = _19_ | a[3];
  assign _21_ = _01_ ^ _11_;
  assign _22_ = _20_ ? _09_ : _21_;
  assign _23_ = _22_ ^ a[3];
  assign _24_ = _20_ ? _12_ : _08_;
  assign _25_ = ~(_24_ ^ a[2]);
  assign _26_ = _23_ & ~(_25_);
  assign _27_ = a[0] & ~(_20_);
  assign _28_ = _27_ ^ a[1];
  assign _29_ = _28_ | _04_;
  assign _30_ = _29_ | ~(_26_);
  assign _31_ = _03_ & ~(_22_);
  assign _32_ = _24_ | a[2];
  assign _33_ = _23_ & ~(_32_);
  assign _34_ = _33_ | _31_;
  assign _35_ = a[1] & ~(_27_);
  assign _36_ = _26_ & ~(_35_);
  assign _37_ = _36_ | _34_;
  assign _38_ = _30_ & ~(_37_);
  assign _39_ = _06_ | ~(_38_);
  assign _40_ = _39_ | ~(_20_);
  assign y[0] = _02_ & ~(_40_);
  assign _41_ = _38_ | _06_;
  assign _42_ = _41_ | ~(_20_);
  assign y[1] = _02_ & ~(_42_);
  assign _43_ = _39_ | _20_;
  assign y[2] = _02_ & ~(_43_);
  assign _44_ = _41_ | _20_;
  assign y[3] = _02_ & ~(_44_);
endmodule

nakengelhardt, 2023-03-20:
X is only meaningful in simulation, it cannot exist in actual hardware. `synth` is a synthesis flow, so it will not preserve X semantics. It assumes that it can make optimizations that will replace undefined values with any value.

For what purpose do you need to have X values preserved after synthesis? There might be some options to get this information.

jix, 2023-03-22:
If you want to compare pre-synthesis (gold) and post-synthesis (gate) results, one approach is to treat `x` on the gold side as matching any value on the gate side (but not the other way around). This strategy is supported by various Yosys builtin commands, e.g. `miter -equiv -ignore_gold_x gold_module gate_module miter_module` pass generates a module `miter_module` that instantiates `gold_module` and `gate_module` on the same inputs and performs such a comparison on the outputs. Note that whatever you do with `miter_module` then still needs to support operations including `x`, e.g. the builtin `sat` command needs a `-enable_undef` option.

Recent versions of Yosys also have an `xprop` pass that can emulate x-propagation by generating extra logic to compute which bits are x. This allows using tools that do not support or do not preserve x-bits to work on something like the output the above miter command. The `xprop` pass itself is still a work in progress, though, and might fail with an error for some designs or leave parts as-is with a warning for other designs.

Related to this, we're also currently in the progress of releasing a new tool, called EQY, which is aimed specifically at equivalence checks like this, in particular for larger designs. While using builtin commands like `miter` and `sat` should already work for such small designs, EQY would still take care of assembling the correct Yosys script for this.

closed 2026-09-28 by ShinyKate