YosysHQ/yosys #3698: Error while running the ibex controller example

Eyantra698Sumanto, 2023-03-08:
### Version

 Yosys 0.26

### On which OS did this happen?

Linux

### Reproduction Steps

Run `make "DESIGN_CONFIG=./designs/asap7/ibex/config.mk"` in the OpenROAD flo scripts.

### Expected Behavior

Yosys should be able to parse all the lines in the Ibex Controller properly.

### Actual Behavior

Yosys is not able to parse the `function automatic [5:0] sv2v_cast_6; ` line 311 of the ibex controller. 

[image]

jix, 2023-03-08:
This syntax is supported starting with yosys 0.8 (released over 4 years ago) and works fine in yosys 0.13 as well as in current releases. It looks like the Makefile is not using the intended yosys version, but a very old version you might have installed separately. I'm not familiar with OpenROAD, but I would guess that you're missing some step to setup your environment so that OpenROAD's yosys is used.

closed 2023-03-08 by jix

Eyantra698Sumanto, 2023-03-08:
Okay. Thanks a lot!